opencomputeproject/onie #121: isohybrid.pl not found -- when building KVM iso

cobracmder, 2015-01-19:
When building kvm_x86_64 in a CI tool using ```build-onie.py -b kvm_x86_64```, the ISO creation process is aborted with the following error message:
```
==== Create kvm_x86_64-r0 ONIE Recovery Hybrid iso ====
I: -input-charset not specified, using utf-8 (detected in locale settings)
Size of boot image is 4 sectors -> No emulation
 58.51% done, estimate finish Mon Jan 19 23:42:26 2015
Total translation table size: 2048
Total rockridge attributes bytes: 1157
Total directory bytes: 4096
Path table size(bytes): 34
Max brk space used 23000
8561 extents written (16 MB)
bash: isohybrid.pl: command not found
make: *** [/home/ubuntu/working/onie/build/kvm_x86_64-r0/stamp/recovery-iso] Err
or 127
``` 

```isohybrid.pl``` script is included in the ```syslinux``` package and is present in ```$ONIE/build/kvm_x86_64-r0/syslinux/syslinux-4.07/utils/isohybrid.pl```.  Will increase the verbosity of ```make``` to see if this is a path issue.

cobracmder, 2015-01-20:
performing a traditional ```cd build-config; make --debug=v MACHINE=kvm_x86_64 all``` results in the same error.

cobracmder, 2015-01-20:
Almost convinced that this is an environment issue in this build VM.  Will continue to dig through this.

cobracmder, 2015-01-20:
Found problem.  Package ```syslinux``` needs to be installed as part of the ```make debian-prepare-build-host``` since it provides ```isohybrid```.  Package ```syslinux-common``` is a dependency of ```syslinux``` and this script provides the ```isohybrid.pl``` script.

Will open a pull request to correct this.

cbrune, 2015-01-20:
Sounds good!

cbrune, 2015-01-20:
closed by:

- a87fd1e Ensure syslinux is installed

cbrune, 2015-01-20:
a87fd1e is reverted.  See discussion in #123

cbrune, 2015-01-20:
Fixed by:

- bc4a13f Qualify isohybrid.pl to be from built syslinux

closed 2015-01-20 by cbrune