opencomputeproject/onie #126: Testing: test case 10 (DHCP Exact - TFTP Server IP + TFTP Bootfile) doesn't pick up file

cobracmder, 2015-02-13:
Using ```./test-onie.py -I eth1 -i 192.168.1.100/24 -m 7072cfaa34fa -t 10 -o onie_installer demo-installer-powerpc-accton_as4600_54t-r0.bin``` results in ONIE not finding the image.

cobracmder, 2015-02-13:
Pcap file: http://1drv.ms/1AxG76P

Based on the pcap file, this is a bug in either the device or ONIE.  Will know more when another device is used.

cbrune, 2015-02-14:
This is a good find!  The log files and the PCAP make this *super* easy to debug.  Thanks!!

The problem is in ONIE and the difference between the BOOTP "file" field and DHCP option 67 "Boot file".

For bootp:
https://tools.ietf.org/html/rfc951

For DHCP options:
https://tools.ietf.org/html/rfc2132

Busybox's udhcpd translates the BOOTP "file" field into a variable named "boot_file", which prepends with "onie_disco_" to finally have: onie_disco_boot_file.  You can see this variable set in the log file.  Also the PCAP shows the "Boot file name" field in the BOOTP section.

The DHCP option 67 is translated into "bootfile" (without the underscore), which gives the ONIE variable onie_disco_bootfile.

The problem is the ONIE code is only using $onie_disco_bootfile .  ONIE should be checking both.

cbrune, 2015-02-14:
Please try this fix in PR #130

cobracmder, 2015-02-17:
From the UTSA reports, PR #130 looks good.

cbrune, 2015-02-17:
fixed by 327ecca

closed 2015-02-17 by cbrune