opencomputeproject/onie #132: testing: test cases  11 & 47 (DHCP Exact - TFTP Server name + TFTP Bootfile) fail

cobracmder, 2015-02-17:
```
./test-onie.py -i 192.168.1.100/24 -I eth1 -m 7072cfaa34fa -t 47 -o onie_updater onie-updater-powerpc-accton_as4600_54t-r0.bin

2015-02-15T16:10:36 - INFO - Loaded 80 tests from config/onie-tests.json
2015-02-15T16:10:36 - INFO - Configuring for Test 47 - ONIE Update - DHCP Exact - TFTP Server name + TFTP Bootfile
Traceback (most recent call last):
  File "./test-onie.py", line 685, in <module>
    main()
  File "./test-onie.py", line 653, in main
    test_args = configure_test(args)
  File "./test-onie.py", line 472, in configure_test
    prepare_test_case(test_args)
  File "./test-onie.py", line 377, in prepare_test_case
    hostname = test_args['local_name']
KeyError: 'local_name'
```

cobracmder, 2015-02-17:
Looks like a simple typo ```local_name``` vs ```host_local_name```.

cbrune, 2015-03-03:
closed by 702c194 and 396d4b8

closed 2015-03-03 by cbrune