opencomputeproject/onie #142: Install GRUB in Embed ONIE fails

Karel-van-de-Plassche, 2015-02-27:
I am trying to port ONIE to a Congatec QA3 board. I've copied the files from ```/machine/kvm_x86_64``` and replaced ```machine/QA3/kernel/config``` by our kernel configs. I can boot ONIE into rescue mode using an USB stick made with ```onie-recovery-x86_64-QA3-r0.iso```, however when I try to run ```onie-self-update -e tftp://xxx.xxx.xx.xx/onie-updater``` I get:
```
ERROR: grub-install failed on: /dev/mmcblk0
Installing for i386-pc platform.
grub-install: warning: File system `ext2' doesn't support embedding.
grub-install: error: embedding is not possible, but this is required for cross-disk install.
```

Maybe related, somehow the install path I defined in ```machine/QA3/install.conf``` gets a 2 appended /dev/mmcblk0 --> /dev/mmcblk02.

cbrune, 2015-02-27:
This issue is being actively discussed here:

  http://lists.opencompute.org/pipermail/opencompute-onie/2015-February/000418.html

Karel-van-de-Plassche, 2015-03-04:
In the above mentioned conversation, we've concluded that the issue was that ONIE only supports /dev/sd* and not /dev/mmcblk* devices. I've created a pull request to add this support: https://github.com/opencomputeproject/onie/pull/144

Karel-van-de-Plassche, 2015-03-16:
This issue is fixed by https://github.com/opencomputeproject/onie/commit/cb129862c114655dc9ba7e1624e9b448fc60606a. This issue can be closed

closed 2015-03-16 by cbrune