opencomputeproject/onie #160: Demo NOS Installation fails (/bin/exec_installer: line 381: /installer: not found)

githubuzer, 2015-03-14:
Hello,

I cloned the ONIE source this week and built the binaries according to the documentation.  Next, I built the Demo OS installer (demo-installer-x86_64-kvm_x86_64-r0.bin) and placed it on my tftp server.  I then booted into ONIE (after selecting Embed) and ran onie-nos-install tftp:/xxx.xxx.xxx.xxx/demo-installer-x86_64-kvm_x86_64-r0.bin.  It fetches the installer and then fails with the above message.  Any suggestions?

Greg

cbrune, 2015-03-16:
Ask for help on the ONIE mailing list.  There are more eyeballs there for this kind of problem:

http://lists.opencompute.org/mailman/listinfo/opencompute-onie

cbrune, 2015-04-15:
is this still an issue? will close in 7 days if no comments.

closed 2015-04-22 by cbrune