opencomputeproject/onie #163: Debian Jessie requires libtool-bin package

cobracmder, 2015-03-22:
In Debian Jessie, *libtool* package (which used to include ```libtoolize``` and ```libtool``` along with aclocal macros) was broken up to have a separate package called libtool-bin which now contains ```libtool```.  This means in Jessie, we need to install both *libtool* and *libtool-bin* packages.

What's the best way to handle this as having *libtool-bin* in the required packages will cause an error during the *debian-prepare-build-host* target?

cbrune, 2015-03-23:
We could release note it somewhere or otherwise put it in a FAQ or known issues document.

My commitment is to make ONIE work with the current stable release of Debian.  Since jessie is not the stable release yet there is nothing to "fix" in ONIE as of today.

cobracmder, 2015-03-23:
No problem.

Would you suggest I make an entry in https://github.com/opencomputeproject/onie/blob/master/docs/faq/index.rst ?

cbrune, 2015-03-23:
I'm thinking we just leave this issue open and that is the "known issue" list.

When jessie is released, we fix ONIE and close this issue.

Until then let this open issue be the tracker.