opencomputeproject/onie #174: Faster/Obvious/Easier abort of ONIE install on X86

cbrune, 2015-04-15:
An issue from the field:

While playing with GRUB on X86, I discovered a couple of unpleasant things, when combined cause a real headache/nightmare:

1) After selecting 'ONIE' in GRUB, it defaults to ONIE Install mode and before you can read all the text the first time (i.e. 4 seconds), it executes ONIE installer

2) ONIE installer loads. Then there isn't a fast easy way (at least no instructions on the screen...anxiety rising..) if any exists, to abort the autoloading/install of the default image via the waterfall (onie-installer-x86_64 found very quickly via the local webserver)

3) While unsuccessfully trying to stop the ONIE install, you watch with horror as ONIE downloads and installs a new OS, overwriting the existing setup. Again, no way I could figure out how to stop that either....

Net result, you have whatever default image is configured on your webserver, and all vestiges of the config and license on your functioning switch has now been erased and must be restored from backup.. you did create a backup.. right?

I would propose that in GRUB, when you go to the ONIE page, that GRUB not be set to autoexecute ONIE Installer in 3 seconds.

I would further propose that an abort sequence be accessible and printed to the screen when ONIE Install mode starts that would allow halting the install for at least 10 seconds after ONIE loads.

mandeepsandhu, 2015-04-15:
+1 to this. I have seen this happen in our lab (thankfully it was only a test setup).

Maybe we can default to the _least-destructive_ `Rescue` mode for the ONIE GRUB menu once it's installed on the system?

Also, once the installer has started, there's no reliable way to stop it right? The installer might be changing the partition table and an abort mid-way might render the box un-bootable.

Or do you mean to say that ONIE would give around 10 secs _before_ executing the NOS installer? This would be definitely helpful as it gives us a window to abort the install (in which case it drops to `ONIE` shell?)

cbrune, 2015-04-16:
The poster makes some valid points.

### Default to `rescue` instead of `install`

I think it *may* be possible to default  to 'rescue' mode.  I say *may* because the default behaviour out of the box from the factory absolutely has to be 'install' mode.  That is the key to the entire automation story.

Now for what happens when a user is manually going through the GRUB menus is a different story.

I can't say off the top of my head, but it seems like we could:

- change the order of the menu items, putting `rescue` first
- at ONIE install time (not OS install time) make sure `install` is set in the grubenv
- examine places where the boot order is set, `onie-boot-mode`.  Look for scenarios were assumptions may have been made about `install` being the default if nothing was selected.  Those places would need to changed to explicitly set `install` mode.

Seems doable, but someone would need to do the work and shake out the details.

### Fast, easy way to stop installer

This seems doable.  When booting into `install` mode the system could:

- print a message about how to turn off install mode -- running `onie-discovery-stop` would do it.
- print the above message, along with a 10 second count down, once a second
- after the count down the system would start the image discovery process for real

That could look like:
```
booting ....
ONIE: OS Install Mode ...
Version   : uefi-x86-201504151848-dirty
Build Date: 2015-04-15T18:48-0700
Info: Mounting kernel filesystems... done.
Info: Mounting ONIE-BOOT on /mnt/onie-boot ...
Running demonstration platform init pre_arch routines...
Running demonstration platform init post_arch routines...
Info: Using eth0 MAC address: 52:54:00:13:34:57
Info: eth0:  Checking link... up.
Info: Trying DHCPv4 on interface: eth0
ONIE: Using DHCPv4 addr: eth0: 10.0.2.15 / 255.255.255.0
Starting: dropbear ssh daemon...random: dropbear urandom read with 6 bits of entropy available
 done.
Starting: telnetd... done.
discover: installer mode detected.  Running installer.
Please press Enter to activate this console.

[Starting installer in 10:  To halt install: Press <Enter> and run `onie-discovery-stop' ]
[Starting installer in 9:  To halt install: Press <Enter> and run `onie-discovery-stop' ]
[Starting installer in 8:  To halt install: Press <Enter> and run `onie-discovery-stop' ]
[Starting installer in 7:  To halt install: Press <Enter> and run `onie-discovery-stop' ]
...
[Starting installer in 2:  To halt install: Press <Enter> and run `onie-discovery-stop' ]
[Starting installer in 1:  To halt install: Press <Enter> and run `onie-discovery-stop' ]
[ Running image discovery ...]
```
Once image discovery starts then it is off to the races.

Of course the actual countdown time could be larger or shorter.  And it only helps if you are connected to the console (or ssh / telnet into the box).

nicolasochem, 2017-08-09:
Hi @cbrune.
This discussion seems to be stalled, but I also have usability reservations about the fact that once you select ONIE at the grub boot stage, you pretty much have to go through the whole reinstall.
The solutions proposed above make sense but seem impractical to implement. Instead, how about an "are you sure" dialog box when you select ONIE install, along with a warning : "Past this point you must proceed to reinstall"
At first view, that would require moving from GRUB to GRUB2 which gives you more freedom with respect to menu/dialogs etc.

cbrune, 2018-08-06:
Closing as several usability issue were fixed with cf6e91c ("NOS mode: a persistent NOS mode boot option")

closed 2018-08-06 by cbrune