opencomputeproject/onie #175: [ONIE-TLV issue] About TLV, type code 0xFD, Vendor Extension.

vic1231, 2015-04-17:
https://github.com/opencomputeproject/onie/wiki/Design-Spec-HW-Requirements

The define says that length of type code 0xFD is Variable. And the value field is a four byte IANA enterprise number, followed by a vendor defined string of bytes.
But after test with onie-syseeprom command with "onie-syseeprom -s 0xFD=1234AAA". The error message says ""ERROR: The value specified is greater than 255". The code takes the value which I want to setting as the byte numbers?

vic1231, 2015-04-17:
Can anyone give me an example to illustrate the type code 0xFD's value field format?

cbrune, 2015-04-20:
Indeed, the documentation for setting type code 0xFD "Vendor Extension" is absent.  Also it is not entirely clear from the source code.  The docs for the command are here:
  https://github.com/opencomputeproject/onie/wiki/Design-Spec-x86-Linux-Kernel

Nevertheless, here is how it works, using a DELL-S6000 as an example:

```
ONIE:/ # onie-syseeprom -s 0xfd="0x12 0x34 0xAA 0x0A"
Adding   TLV 0xfd: Vendor Extension
Programming passed.
TlvInfo Header:
   Id String:    TlvInfo
   Version:      1
   Total Length: 95
TLV Name             Code Len Value
-------------------- ---- --- -----
Service Tag          0x2F   7 2RY8VS1
Base MAC Address     0x24   6 44:38:39:00:7B:0E
MAC Addresses        0x2A   2 129
Product Name         0x21   5 S6000
Manufacturer         0x2B   4 DELL
Serial Number        0x23   7 2RY8VS1
Platform Name        0x28  26 x86_64-dell_s6000_s1220-r0
Loader Version       0x29  10 2015.02.01
Vendor Extension     0xFD   4  0x12 0x34 0xAA 0x0A
CRC-32               0xFE   4 0x7CA831DF
Checksum is valid.
```

There a few particularities you have to watch out for:

- The type code must be `0xfd` (lower case).  Using `0xFD` will not work.  That should be fixed.
- The value is a *space* separated list of bytes.
- The bytes can be specified in decimal or hex.  If hex then the `0x` prefix is required.

All of this should end up in the docs for this command.

vic1231, 2015-04-21:
Dear cbrune:
Thanks for your clear explain and illustrate. I will take care with "space separated list of bytes" and make a try on newest ONIE github code. And maybe fixed the case issue(0xfd, 0xFD).

closed 2015-04-21 by vic1231