opencomputeproject/onie #177: [OCE (ONIE Compliance Environment) question]

vic1231, 2015-04-22:
In
https://github.com/opencomputeproject/onie/blob/master/contrib/autobuild/README.md

On the [Testing ONIE] section, I followed the step, including 
1. sudo service apparmor teardown
2. sudo apt-get install python-pip python-virtualenv virtualenv .venv
3. source .venv/bin/activate
4. pip install -r requirements.txt
and last, I key in "./test-onie.py -c config/onie-tests.json -d config/dut-example.json -t 69 -I onie0" for a test.
The screen message is like below:
-------------------------------------------
2015-04-22T18:26:21 - INFO - Loaded 80 tests from config/onie-tests.json
2015-04-22T18:26:21 - INFO - Loaded DUT config from config/dut-example.json
2015-04-22T18:26:21 - INFO - Configuring for Test 69 - ONIE Update - TFTP Waterfall - Name 1
2015-04-22T18:26:21 - INFO - Starting tftp
2015-04-22T18:26:21 - INFO - Starting dhcp

Press enter to stop all services
-------------------------------------------
I think maybe this test method is work.

But my question is ....Is some action should I do in KVM (ONIE DUT) so that let me pass the test 69?
And it will write some msgs in output/test-69?

The config "mac_address" in dut-example.json I think it is DUT's MAC address.

Are there anyone can help me or show me some info links to understand about this?

cobracmder, 2015-04-22:
Hey @vic1231.

At the last line on the screen,
```
Press enter to stop all services
```
You DUT (ONIE VM) is ready to be tested using the ONIE Self Update mechanism with the TFTP Waterfall - Name 1 naming scheme.

Take a look at ```output/test-69/tftproot```.  You'll see your onie updater image was renamed to the proper naming scheme.  At this point, if you select ```ONIE Update``` from GRUB or withing ONIE itself, the DUT will proceed through image discovery and should find the update image via TFTP Waterfall on the given name.

There are some 'issues' (I call them bugs) on some daemons like ```dhcpd``` which do not log their output to a given log file (even when you specify it to), instead it logs to the default syslog target ```/var/log/messages``` or ```/var/log/syslog``` depending on system configuration.  You should see the dhcpd output there.  This is also why during proper certification testing, we record all console interaction from the DUT (since we can't fix system daemons).

The item for success is, did the DUT perform the action (update) on the given mechanism (tftp) using the proper naming scheme (Name 1).  If it did, test is a success.

As per the DUT configuration, take a look at https://github.com/opencomputeproject/onie/blob/master/contrib/autobuild/config/dut-example.json and the explanation of the file https://github.com/opencomputeproject/onie/blob/master/contrib/autobuild/docs/dut-config.md .  The ```mac_address``` field is the MAC address for the DUT.

In the DUT configuration, you can also specify additional options (check out ```test-onie.py -h``` for all supported options).  This allows for specifying them per DUT vs on the command line all of the time.  Check out: https://github.com/opencomputeproject/onie/blob/master/contrib/autobuild/config/accton_as4600_54t.json as an example.

Let me know if you have any questions.

vic1231, 2015-04-23:
Dear cobracmder:
I filled the correct value in dut-example.json. It looks like:
{
    "name": "DUT Configuration",
    "mac_address": "00:00:00:00:00:01",
    "ip_cidr": "192.168.1.3/24",
    "options": {
        "onie_machine": "x86_64",
        "onie_machine_rev": 0,
        "onie_vendor": "kvm",
        "onie_arch": "x86_64",
        "onie_installer": "onie-installer-x86_64-kvm_x86_64-r0",
        "onie_updater": "onie-updater-x86_64-kvm_x86_64-r0"
    } 
}
After I enter command 
"./test-onie.py -c config/onie-tests.json -d config/kvm_x86_64.json -t 69 -I eth0", and path autobuild/output/test-69/tftp-root have the updater file, ex. "onie-updater-x86_64-kvm_x86_64-r0".

Now my DUT (ONIE VM) can get the updater and install ONIE success. But it still get updater file from my own tftp folder, like /tftpboot, not autobuild/output/test-69/tftp-root. So, the log file in test-69/logs/tftpdhpa.log does not have any message.

Should I modify my tftp path from /tftpboot to autobuild/output/test-69/tftp-root?
Or something else I missed?

Regards,

cobracmder, 2015-04-23:
Hey @vic1231,  I've ran into this issue before.

When you install ```isc-dhcp-server```, ```tftpd-hpa``, and ```nginx```, they will be configured to autostart when the system is up.  However, we need to ensure they are disabled as OCE needs to control them.

If you are on a Debian based system, ```update-rc.d``` is your friend: http://manpages.ubuntu.com/manpages/hardy/man8/update-rc.d.8.html.

That will take care of ```dhcpd```, ```nginx```, and ```tftpd-hpa```, now let's turn our attention to ```tftpd```.  ```tftpd``` is usually not started standalone but rather through ```inetd``` or ```xinetd```.  

If ```xinetd``` is being used, check out ```/etc/xinetd.d``` for a file called ```tftp```.  Edit that file to make ```disable = yes``` and restart ```xinetd```.

If ```inetd``` is being used, check out ```/etc/inetd.conf```, comment out the line(s) for tftpd, and restart ```inetd```.

To ensure all of those services are disabled, you can run:
```
# sudo netstat -atupn
```
This will list all active and listening TCP and UDP ports while displaying which PID/Process is bound to the port.

If this doesn't sort you out, ping me directly and we do a BlueJeans session to triage.

vic1231, 2015-04-24:
Dear cobracmder:
Sure, you're right. First, there always have tftp process there:
++++++++++++++
sudo netstat -atupn | grep ftp
udp        0      0 0.0.0.0:69              0.0.0.0:*                           4148/in.tftpd
++++++++++++++
After I edit /etc/xinetd.d/tftp and modify disable to "yes", and enter two commands:
sudo service xinetd stop
sudo service tftpd-hpa stop
Then tftp service really stop/waiting, so that tftp service can run again via execute test-onie.py.
Thus, the tftp server path is chang to /contrib/autobuild/output/test-69/tftp-root.

For make sure, I changed file name in /test-69/tftp-root from "onie-updater-x86_64-kvm_x86_64-r0" to "onie-updater-x86_64-kvm_x86_64-r00", and  my DUT (ONIE VM) can't not get file to install via tftp. Obviously, it is make sense, and after I changed the file name from "onie-updater-x86_64-kvm_x86_64-r00" to "onie-updater-x86_64-kvm_x86_64-r0", then my DUT download file and install successfully.

But there is still no msg records in /test-69/log/tftpdhpa.log, is this a problem like dhcpd issue?
I can use some tftp logs from /var/log/syslog. Or maybe it's another issue.
At least, my DUT can download file form tftp server which test-onie.py triggered.
------------------------------------------------------------------------

Furthermore, I found some test items had not implement yet with  test-onie.py, like test 0 and 1.
Include test 38-43, the part of Static Loader and USB-Names, also test item 74-79?

cobracmder, 2015-04-24:
Hey @vic1231,

It sounds like ```tftpd-hpa``` is also using the built-in syslog system so you should check your ```/var/log/messages``` or ```/var/log/syslog```.

You are correct, OCE doesn't have tests that require manipulating the console and / or power cycling.  Those are still required to be done manually.  If you would like to contribute to OCE by writing the console manipulation or power cycling piece(s), that would be much appreciated.

The user should be able to uninstall a NOS by using one of the following areas:
* Boot loader (GRUB on x86_64 or UBoot every where else)
* ONIE (in Rescue mode)
* From the NOS itself

Uninstalling from the bootloader is easy:
On ```x86_64``` select ```ONIE Uninstall``` from GRUB.
Everywhere else, on the U-Boot prompt, ```run onie_uninstall```

Uninstalling from ONIE rescue mode means enter ONIE rescue mode, and perform ```onie-uninstall``` which is essentially the commands here: https://github.com/opencomputeproject/onie/wiki/Design-Spec-x86-NOS-Interface#x86-nos-uninstall.

Uninstalling from the NOS (Demo OS in this case), is running the ```uninstall``` script in the Demo OS.

At no point should ONIE be removed from the device.

For Test 79, that is correct.  We are ensuring the NOS is able to set ONIE into one of the given states so that test passed.

vic1231, 2015-04-26:
Dear cobracmder:
Thanks for your information, and I'll try that!

Regards,

closed 2015-04-29 by vic1231