opencomputeproject/onie #178: [ONIE Certification] ONIE-test-form.xlsx, DUT testing information values.

vic1231, 2015-04-24:
Some ONIE testing information I found in
http://www.opencompute.org/wiki/Networking/ONIE/Testing%2BCertification
The testing report link http://files.opencompute.org/oc/public.php?service=files&t=c81b4a02ad7d1b5b6de16ca2935fa3cc

There some columns should be filled but not about ONIE actions. I listed in below:
1.    Vendor
2.    Model
3.    Serial Number
4.    MAC Address
5.    Switch Label Present	
6.    Certification Lab	
7.    Testing Date	
8.    ONIE Version on Switch
9.    ONIE Version for Certification
10.  ONIE image freely available on Web
11.  ONIE model files available ONIE GitHub
12.  USB Port on Device
13.  Test Status
I know some column can easily fill value by use command onie-syseeprom to get informations.
For "Vendor" column, we can use onie-syseeprom -g 0x2D to getting values in TlvInfo EEPROM.

But for make sure, I want to ask about is there someone knows exactly TlvInfo EEPROM <-->column matching?
For example, I think the first column "Vendor" is matching TlvInfo EEPROM type code "0x2D". And more:
2.   Model	: 0x21
3.   Serial Number: 0x23
4.   MAC Address:  0x24
8.    ONIE Version on Switch: 0x29
9.    ONIE Version for Certification: 0x29

I want to meet someone's opinion as a double check.

cobracmder, 2015-04-24:
Hey @vic1231,

In the ONIE Test Report, there are high level fields that the certification lab fills out.  This is done using the information provided from the vendor and confirmed by the ```onie-syseeprom``` command and other pieces of data found on the box.

### Vendor
This is the vendor supplying the device to consumers / end users.

### Model
This is the model number of the device.

### Serial Number
This is the serial number of the device.

### MAC Address
This is the MAC address of the management port.

### Switch Label Present
This is a boolean field to indicate if the device meets the Asset Tracking requirements: https://github.com/opencomputeproject/onie/wiki/Design-Spec-HW-Requirements#asset-tracking-and-labeling.

### Certification Lab
This is a string to indicate which Certification Lab performed this assessment.

### Testing Date
This is a date to indicate when the Certification Lab performed the assessment.

### ONIE Version on Switch
This is the version string the vendor uses to represent their version of ONIE (e.g. 2014.08.03-01).  This version string should include the base version of ONIE (2014.08 in this example) and the vendor is able to use their build / QA / configuration management scheme with it.

### ONIE Version for Certification
This is the version of ONIE the vendor is based against.  In the previous example, this would be 2014.08.

### ONIE Image Freely Available on Web
This is a boolean field to indicate if the images for the device are freely available from the vendors website without being behind a pay wall or requiring a login: https://github.com/opencomputeproject/onie/wiki/Certification-Proceedure#prerequisites

### ONIE model files Available on ONIE GitHub
This is a boolean field to indicate if the ONIE device code is available in the ONIE respository for that particular release.  https://github.com/opencomputeproject/onie/wiki/Certification-Proceedure#prerequisites

### USB Port on Device
This is a boolean field to indicate if the device has a USB device.  If not, all USB tests will marked 'N/A'.

### Test Status
This is a highlevel test indicator made up from the previous requirements and the 80 test cases.

vic1231, 2015-04-26:
Dear cobracmder:
Thanks for your information, it's really detailed.
I'll keep groping with this funny project.

vic1231, 2015-04-28:
Dear cobracmder:
Do you know below test item:
1. ONIE Image Freely Available on Web, and 
2. ONIE model files Available on ONIE GitHub
in high level fields of  the ONIE Test Report is necessary and should implement for pass certification?
 
--- Is it necessary to put an ONIE image which we built on our public server for download test?
--- And should I upload codes to GitHub with what I modified, then get a tag or trunk log, etc, and just can pass" ONIE model files Available on ONIE GitHub" test item?

I mean, these field just only have "1" (assume it means yes, or Available) answer?
If got "0", the certification failed?

Regards,

cobracmder, 2015-04-28:
Hey @vic1231,

They are prerequisites for ONIE Compliance Testing (https://github.com/opencomputeproject/onie/wiki/Certification-Proceedure#prerequisites).  If these are not done, it doesn't matter if a vendor passes all tests or not, it is not granted certification.

If a vendor is distributing a device with ONIE, they must be able to provide their customers with bug fixes and feature improvements.  ONIE provides for this with update images.  It is these images that need to be available.

A vendor's ONIE implementation should be publicly available.  The vendor may maintain their own repo like what Plexxi has done (https://github.com/Plexxi/onie) but the preferred method is to have the platform files in the main ONIE repo (https://github.com/opencomputeproject/onie) like what: Accton, Alpha Networks, Celestica (RedStone only so far), Centec, Dell, Mellanox, and Quanta do today.

I'll use Accton as the example of how they do their ONIE development:
* Once they have prototype hardware for a device, they port ONIE to it using a release branch or the master branch
* Once they are happy with bringing up ONIE and the box is happy, they perform a Pull Request to the main ONIE repo for their given platform (https://github.com/opencomputeproject/onie/commit/f4515070aa465f963a2fc6e7de9a9a846ca2452a#diff-ad90a4bedeb83fa75efaa1240c104a96)
* As bug fixes are found, they are validated and a Pull Request is made to the main ONIE repo
   * Sometimes these bugs are only for their platform, sometimes it's a bug that affects everyone
* When an ONIE release occurs (https://github.com/opencomputeproject/onie/wiki/Release-Cycle), these vendors may choose to move their platforms to that current release.  If they do, they will build ONIE from that release branch, perform testing and QA, perform a release to their customers (ONIE images to website, release notes, customer communication, etc...), and submit for ONIE Certification for that release.
* Now the cycle repeats

Let me know if this helps.

vic1231, 2015-04-29:
Dear cobracmder:
Your information is helpful for me. If I have product code to be checked out or release, I'll pull a Request. 

Now I' trying to create my own branch for practice and testing Git utilities. Like "vic1231/onie"
But it's a little inconvenient if I want to create /machine/test/file_1 to /machine/test/file_10 for an example. I will press 1 time with "+" sign(Create a new file here) for create test/, and press 10 times "+" sign for create file_1 to file_10. I do copy my code or configs context from my local files that I have changed from branch(ex, 2015.05-rc1), and paste them to webpage, then press "Commit new file" button. I think there is no problem to create file/directory.

But I'm trying to find out if there are any easier way to add file/directory in my ONIE branch, maybe use command line to add file/directory if it exist.
Or, maybe this is the best way now to use webpage add file/directory.
Do you know which way is better or easier?

cobracmder, 2015-04-29:
Hey @vic1231,

For starters, take a look at:
* https://github.com/opencomputeproject/onie/wiki/Contributing

And if you're new to github:
* https://help.github.com/articles/fork-a-repo/
* https://help.github.com/articles/creating-and-deleting-branches-within-your-repository/#creating-a-branch
* https://help.github.com/articles/creating-a-pull-request/

As you've mentioned already, using the GitHub web interface to achieve most of this can be quite cumbersome.  As such, I would recommend using the CLI for as much as possible.

For example:
* Fork onie repo into your account
* ```git clone https://github.com/vic1231/onie```
* ```git checkout -b **BRANCH_NAME** ``` # change BRANCH_NAME to be the topic, like pebble
* Add your code (files and dirs)
* Make sure your code works on a set of boxes (if you are adding a new platform, make sure it works on your box)
* ```git add **FILES**``` # stage files for a commit
* ```git commit``` # fill out the commit message and commit to your local repo
* ```git push``` # push your new branch to your github account
* From the GitHub interface, navigate to https://github.com/vic1231/onie and select 'Create Pull Request'

Once you get comfortable with the git CLI, you might want to check out Hub (https://hub.github.com/) which is a wrapper around git that adds GitHub capabilities into git.

Let me know if this helps.

vic1231, 2015-04-30:
Dear cobracmder:
Sure, these information is helpful, and some of links I didn't noticed before. 
Your example is a quick start way. After few times of doing these steps, I take a pull request for test(my branch name now is vic_onie_4). I found that there are files under machine/ folder which I committed(ex, machine/xxx_x86_64/). But when I use command "git clone https://github.com/vic1231/onie.git" to download my ONIE branch again, the files which I committed isn't exist. Should I waiting for pull request permitted?

I take ONIE master branch as my orange ONIE(vic1231/onie), then create branch vic_onie_4 from vic1231 master branch.
And now I click "pull request button" with my branch vic_onie_4.
If the pull request permitted, then it just matters with vic1231/onie, not about ONIE master, right? And if my commit want to add to ONIE master branch, it must take pull request again but with ONIE master branch?

Now, I am waiting for download my branch and take a build test for check my code and configs again.

Furthermore, do you know how to delete unused branches that I create before?
I create some branches before, like patch-1, vic1231-patch-1, vic_onie_1, ..., etc, but now I just need branch vic_onie_4.

And I will study more with these github informations.

Regards,

cobracmder, 2015-04-30:
Hey @vic1231,

Take a look at https://help.github.com/articles/creating-and-deleting-branches-within-your-repository/#deleting-a-branch for deleting a branch on github.

cbrune, 2015-04-30:
@vic1231 

When you make a "pull request" you are "requesting" that the ONIE project "pull" in your changes.

Your pull request does *not* mean your changes automatically become part of the ONIE repo.

Your proposed changes have to be reviewed by the ONIE project first.  If all the changes look good then we can pull it into the repo.

cbrune, 2015-04-30:
@vic1231 -- also please read https://github.com/opencomputeproject/onie/wiki/Contributing carefully.

That document explains how to contribute your changes to the ONIE project.

cbrune, 2015-04-30:
Hi @vic1231 -- I made a number of comments in #183 

Please review those comments.

For additional questions please use the ONIE mailing list:
http://lists.opencompute.org/mailman/listinfo/opencompute-onie

vic1231, 2015-05-01:
Helo cobracmder:
Thanks for your information, I will take a look.

vic1231, 2015-05-01:
Hello cbrune:
I will check these issues.

vic1231, 2015-05-06:
I'm studying at Testing https://github.com/opencomputeproject/onie/wiki/Testing-Overview
Which introduce Testing Environment and Recommended environment - Linux.
My question is, should I prepare my personal PC or notebook to Lab? And this PC should include ONIE project and testing OCE related files that fill correct parameters? ex, contrib/autobuild/config/model_name.json, contrib/isc-dhcpd/dhcpd.conf.

cobracmder, 2015-05-06:
Hi @vic1231,

The ONIE Certification Lab will have the entire environment ready.  They will build the dut config file when the device(s) are submitted.  All a vendor has to do is provide the devices and ensure all the Prerequisites [https://github.com/opencomputeproject/onie/wiki/Certification-Proceedure] are done before submitting the devices for testing.

cbrune, 2015-05-06:
I would think a HW vendor would make the rather modest investment in standing up their own ONIE testing server in order to do as much 'self testing' as possible before submission.

@cobracmder -- Can the ONIE testing server be a VM to keep cost down?

cobracmder, 2015-05-06:
@cbrune, you bet.

It is advised that a HW vendor would use OCE prior to compliance testing at the lab.  For that, the vendor may find that using a VM to be easy to deploy and use as the ONIE testing server.

vic1231, 2015-05-07:
Dear all:
My test environment is real machine with Ubuntu 14.10.
In OCE test, I passed all the test item exclude some test item should be mutual like test 0-7, or test 77-79.
You solve my issues and give me many advise.
Thanks all of you.

And I think my current task in learning git utility.
If there are questions with ONIE Certification, I will re-open this issue to discuss with everyone.

closed 2015-05-07 by vic1231