opencomputeproject/onie #201: ISO  image still has UEFI boot method when supporting UEFI firmware disabled

david56, 2015-05-29:
Yesterday, I found the ISO image always combined with BIOS and UEFI booting methods even if ``UEFI_ENABLE = no``.  If the recovery ISO which has linux-3.2.25 boots via ``UEFI: JetFlashTrans...``,  the kernel will have some errors and cause eUSB device disappeared.  ONIE updater can't find the mass storage to install itself eventually.

    ONIE: Rescue Mode ...
    Platform  : x86_64-accton_as5712_54x-r0
    Version   : 2015.5.28-dirty
    Build Date: 2015-05-28T11:16+0800
    ehci_hcd 0000:00:16.0: Found HC with no IRQ.  Check BIOS/PCI 0000:00:16.0 setup!
    ehci_hcd 0000:00:16.0: init 0000:00:16.0 fail, -19
    Info: Mounting kernel filesystems... done.
    Info: Using eth0 MAC address: 00:00:e8:a0:00:00
    Info: eth0:  Checking link... up.
    Info: Trying DHCPv4 on interface: eth0
    ONIE: Using DHCPv4 addr: eth0: 192.168.1.230 / 255.255.255.0
    Starting: dropbear ssh daemon... done.
    Starting: telnetd... done.
    discover: Rescue mode detected.  Installer disabled.
    
    Please press Enter to activate this console.
    To check the install status inspect /var/log/onie.log.
    Try this:  tail -f /var/log/onie.log
    
    ** Rescue Mode Enabled **
    ONIE-RECOVERY:/ # ls /sys/firmware/
    memmap
    ONIE-RECOVERY:/ #

``acpi`` device is also absent.

cbrune, 2015-05-29:
I don't quite understand from your description what the problem is.

Can you put the console output from the ONIE install process on a gist so we can all look at it?

Also to aid debugging you can add "set -x" near the top of `installer/x86_64/install-arch` to see more output.

david56, 2015-06-02:
Here you are:

https://gist.github.com/david56/ffedf7ac4d4f32aff0fd

cbrune, 2015-06-05:
I think I understand the situation, but I do not understand why the 3.2.25 kernel cannot find the USB storage.

onie-mk-iso.sh always creates an ISO that can be used from legacy BIOS and from UEFI.  Also you are forcing the UEFI boot method of the ISO from your firmware's menu.  So a non-UEFI ONIE boots up in a UEFI environment....  That really should not be a problem.  The big problem here is the USB devices are not showing up in Linux in this situation....  Why that is I do not understand.

Is `UEFI: JetFlashTrans....` the regular way you have booted ONIE in the past, before all the UEFI changes went in?  Is there a non-UEFI boot option to use when booting the ISO from USB?

My other thought is maybe onie-mk-iso.sh should be smarter and look at `$UEFI_ENABLE` and only include both ISO boot methods in the case of `auto`.  In the case of `bios` only build a MBR ISO.  In the case of 'uefi' only build a UEFI ISO.

david56, 2015-06-10:
> Is UEFI: JetFlashTrans.... the regular way you have booted ONIE in the past, before all the UEFI changes went in? Is there a non-UEFI boot option to use when booting the ISO from USB?

With the new recovery ISO, there are two boot options, ``JetFlashTranscend 8GB 8.07`` and ``UEFI: JetFlashTranscend 8GB 8.07``, in spite of ``UEFI_ENABLE = no``.  In the past, it had only ``JetFlashTranscend 8GB 8.07``.

> My other thought is maybe onie-mk-iso.sh should be smarter and look at $UEFI_ENABLE and only include both ISO boot methods in the case of auto. In the case of bios only build a MBR ISO. In the case of 'uefi' only build a UEFI ISO.

+1

david56, 2015-08-05:
Fixed in commit 4e78020

closed 2015-08-05 by david56