opencomputeproject/onie #208: About NOS image size

vic1231, 2015-06-04:
In installer file, installer/x86_64/install-arch, it defines 128MB as default NOS partition size. It seems enough to store ONIE Demo OS. But if I use another OS(not ONIE demo OS) as a NOS, it's image size may be over 128MB, can I change the NOS partition size(maybe change to 1G, or bigger) in file install-arch?

cbrune, 2015-06-04:
It seems you are talking about the DEMO OS installer.  That configuration is an *example*.  It is to demonstrate how an OS installer interacts with an ONIE enabled system.

An OS vendor takes that as an example and adapts it (incorporates it) into their OS installer, presumably with modifications specific to their OS.

closed 2015-06-04 by cbrune

vic1231, 2015-06-05:
Where can I get some vendor's NOS information, like NOS image/image size, unix/linux brahch(cent os, ubuntu, fedora, etc.) and how it made? I don't have much sense about building a NOS for ONIE installer.