opencomputeproject/onie #214: USB install should be slightly more verbose

dorkmatt, 2015-06-15:
It would be useful if ONIE was slightly more verbose when attempting to mount a USB stick. In the current form, only a successful mount is logged as.. 

````
ONIE: Failed to configure eth0 interface
ONIE: Starting ONIE Service Discovery
ONIE: Executing installer: file://dev/sda1/onie-installer-powerpc
Verifying image checksum ... OK.
Preparing image archive ... OK.
```

However, if the stick is formatted incorrectly (wrong partition type) and/or has the wrong file name, the log doesn't show this as an error. Seems if we have a USB stick present and had issues with it, we should advertise that. Just as we do with eth0 not having link, a DHCP server not being found, etc.

cobracmder, 2015-06-15:
Hey @dorkmatt , this sounds like a good idea.  Can you attach a patch/pr to get a glimpse of how much logging should be done?

cbrune, 2015-06-15:
Just because a USB is present does not mean the user intends to boot from it.  It might be there for various reasons.  Today ONIE does not know it is USB, just that block device partitions are present.

That said we could print some messages about each partition found.  A mock up of the desired output would be a good starting point.

cbrune, 2017-04-17:
This has been duplicated by issue #467.  A pull request for the fix is in #501.

commit daeb0b4f (2017-04-19): Log local file image discovery paths

The http and tftp image discovery mechanisms log helpful information
about the attempted URL paths.  The local file (typically USB)
discovery mechanism, however, does not.  This is a historical
oversight.

This patch adds some logging to the local file discovery mechanism.
The discovery mechanism now logs the disk partition and file name for
each discovery attempt.

As an example, the kvm_x86 machine log for local file discovery now
logs this:

  Info: Attempting file://dev/vda3/onie-installer-x86_64-kvm_x86_64-r0 ...
  Info: Attempting file://dev/vda3/oni [... cut]
closed 2017-04-19 by cbrune