opencomputeproject/onie #220: Update wiki for  image discovery waterfall: add well known server name #219

cobracmder, 2015-07-01:
Cursory search reveals two pages that need updating for #219:
https://github.com/opencomputeproject/onie/wiki/Design-Spec-SW-Image-Discovery
https://github.com/opencomputeproject/onie/wiki/Testing-Overview

cbrune, 2015-07-01:
Mostly done.  Please review.

https://github.com/opencomputeproject/onie/wiki/Design-Spec-SW-Image-Discovery#default-onie-image-server-name-onie-server

https://github.com/opencomputeproject/onie/wiki/Testing-Overview#inexact-urls-based-on-default-onie-server-name-onie-server

For the test cases I was not sure what test numbers to use.

cobracmder, 2015-07-01:
No problem. Since we have some more test cases to add for another issue, I'll take ownership of this case to add the proper case numbers with the test reports.

cobracmder, 2015-07-02:
done.  Added tests 80 to 89 for install and 95 to 104 for update

closed 2015-07-02 by cobracmder