opencomputeproject/onie #223: Update wiki for image discovery waterfall: use RFC-3927 link-local IPv4 address #222

cbrune, 2015-07-01:
Update wiki for image discovery waterfall: use RFC-3927 link-local IPv4 address.  This relates to PR #222.

Cursory search reveals two pages that need updating for #222:
https://github.com/opencomputeproject/onie/wiki/Design-Spec-SW-Image-Discovery
https://github.com/opencomputeproject/onie/wiki/Testing-Overview

cbrune, 2015-07-02:
I updated the SW design spec to include the link-local address scheme.

I also updated the Testing Overview wiki, adding a new section "IPv4 Link Local Address (RFC-3927)", which is analogous to the IPv6 neighbor test section.  I left the test number TBD.

cbrune, 2015-07-02:
Moving to Carlos.  Close when the test numbers are figured out.

cobracmder, 2015-07-02:
done. Added tests 90 to 94 for install and 105 to 109 for update.

closed 2015-07-02 by cobracmder