opencomputeproject/onie #249: 50_onie_grub does not generate correct config

lakesh-sharma, 2015-08-17:
50_onie_grub should be invoked by NOS installer.
-it would re-generate the onie's grub config.
-And it would provide onie's chainload menu entry, along with any available
  diags entry config to NOS grub config.

At present, it seems that 50_onie_grub would not generate correct config entries.
It would miss some of config params listed in machine config.
More on this issue has been discussed in thread, and possible solutions high-lighted.

http://lists.opencompute.org/pipermail/opencompute-onie/2015-August/000826.html

cbrune, 2015-08-28:
See patch sent to mailing list:

http://lists.opencompute.org/pipermail/opencompute-onie/2015-August/000836.html

lakesh-sharma, 2015-08-28:
the patch looks good.
grub-extra.cfg would be generated by build infra ?

cbrune, 2015-08-28:
yes.  grub-extra.cfg is generated by build-config/scripts/onie-mk-installer.sh during final image creation.

closed 2015-08-28 by cbrune

paavaanan, 2020-05-02:
is there any workaround available for this bug?

commit 1225f970 (2015-08-28): grub: extra kernel command line arguments

Fixes #249

Some platforms require extra kernel command line arguments in order to
function properly.  These arguments were not properly carried forward
in all circumstances, such as when installing a NOS.

These extra arguments are specified in a platform's machine.make file
using the $EXTRA_CMDLINE_LINUX variable.

Previously these arguments were dynamically inserted into the ONIE
grub.cfg file using search and replace at image build time.
Unfortunately at *run time* 50_onie_grub script was unaware of these
extra arguments.

This patch moves the ext [... cut]