opencomputeproject/onie #250: x86_64: grub may become invalid if power off during making install boot mode sticky

david56, 2015-08-21:
In recent ONIE install mode, it always reinstall grub and set boot mode to ``install`` when auto discover starting.  We found it may make grub become invalid if power off suddenly.

    Version   : 2015.6.9
    Build Date: 2015-06-09T15:55+0800E: OS Install Mode ...
    Info: Mounting kernel filesystems... done.
    scsi 6:0:0:0: Direct-Access     ATP      ATP IG eUSB      1100 PQ: 0 ANSI: 4
    sd 6:0:0:0: [sda] 15724544 512-byte logical blocks: (8.05 GB/7.49 GiB)
    sd 6:0:0:0: [sda] Write Protect is off
    sd 6:0:0:0: [sda] No Caching mode page present
    sd 6:0:0:0: [sda] Assuming drive cache: write through
    sd 6:0:0:0: [sda] No Caching mode page present
    sd 6:0:0:0: [sda] Assuming drive cache: write through
    sd 6:0:0:0: [sda] No Caching mode page present
    sd 6:0:0:0: [sda] Assuming drive cache: write through
    sd 6:0:0:0: [sda] Attached SCSI disk
    scsi 7:0:0:0: Direct-Access     JetFlash Transcend 8GB    8.07 PQ: 0 ANSI: 4
    sd 7:0:0:0: [sdb] 15261696 512-byte logical blocks: (7.81 GB/7.27 GiB)
    sd 7:0:0:0: [sdb] Write Protect is off
    sd 7:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
    sd 7:0:0:0: [sdb] Attached SCSI removable disk
    Info: Mounting ONIE-BOOT on /mnt/onie-boot ...
    Info: Using eth0 MAC address: 00:00:e8:a0:00:00
    Info: eth0:  Checking link... up.
    Info: Trying DHCPv4 on interface: eth0
    ONIE: Using DHCPv4 addr: eth0: 192.168.1.230 / 255.255.255.0
    Starting: dropbear ssh daemon... done.
    Starting: telnetd... done.
    Installing for i386-pc platform.
    Installation finished. No error reported.
    (power off)

Power on again, the grub becomes invalid.

    error: file `/grub/i386-pc/normal.mod' not found.
    Entering rescue mode...
    grub rescue>

    error: file `/grub/i386-pc/extcmd.mod' not found.
    Entering rescue mode...
    grub rescue>

The problem looks like be inevitable in present grub usage.  Should we add a warning message to look out the risk?

david56, 2015-08-21:
Another grub broken case:

    GRUB loading.
    Welcome to GRUB!
    
    error: invalid arch-independent ELF magic.
    Entering rescue mode...
    grub rescue>

cbrune, 2015-08-21:
The install stickiness is intentional and had always been there for PowerPC.

The legacy BIOS GRUB/MBR method is admittedly clumsy.  It probably has a wider window of "don't turn off here" then other methods.  This window is much shorter in PowerPC (updating u-boot env variables) and UEFI (updating EFI variables).

All systems will be broken for a perfectly timed power off situation.  For u-boot imagine the power goes out while updating an env variable.  The env variable udpate operation requires first erasing the env flash sector and then writing back the updated sector.  If the power goes out after the erase, but before the write then the box will have problems.  u-boot would still be there, so a recovery could proceed using u-boot commands.

Similar comments for a UEFI system -- the UEFI shell could be used to recover.

The legacy BIOS method has the most problems, since when GRUB is broke there is no recovery shell, like u-boot and UEFI have.

I don't know what to do about the power going off at the wrong, outside of minimizing the exposure.  Perhaps in this case the installer stickiness could be handled better, perhaps inspecting things to see if the update-grub operation is really necessary to make install mode sticky for a particular case.

cbrune, 2018-08-06:
closing as old

closed 2018-08-06 by cbrune