opencomputeproject/onie #263: insmod and rmmod in ONIE updater triggered by onie-self-update will run 3 times.

david56, 2015-09-15:
Hello Curt,

Recently, I moved Accton's firmware updater to run on ONIE v2015.08.  And found ``insmod`` and ``rmmod`` in ``sharch_body.sh`` or ``install.sh`` (even ``install-arch``) will execute 3 times if the script triggered by ``onie-self-update``.  i.e., everything is find if running the script via linux shell directly.

I'm looking for the cause but have no idea for what's going on.  I tried to add some code in ONIE updater's ``sharch_body.sh``.  There is the same issue while doing ONIE update.

    diff --git a/installer/sharch_body.sh b/installer/sharch_body.sh
    index cd6d3b4..247d2f7 100644
    --- a/installer/sharch_body.sh
    +++ b/installer/sharch_body.sh
    @@ -82,6 +82,10 @@ fi
         clean_up 1
     }
    
    +echo "doing insmod ..."
    +insmod xyz.ko
    +echo "doing rmmod ..."
    +rmmod xyz
     $tmp_dir/installer/install.sh "$@"
     rc="$?"

Result:

    ONIE:/ # update_url tftp://192.168.1.99/onie-updater-x86_64-accton_as5712_54x-r0
    discover: Rescue mode detected. No discover stopped.
    ONIE: Unable to find 'Serial Number' TLV in EEPROM data.
    Info: Fetching tftp://192.168.1.99/onie-updater-x86_64-accton_as5712_54x-r0 ...
    onie-updater-x86_64- 100% |*******************************| 11431k  0:00:00 ETA
    ONIE: Executing installer: tftp://192.168.1.99/onie-updater-x86_64-accton_as5712_54x-r0
    Verifying image checksum ... OK.
    Preparing image archive ... OK.
    doing insmod ...
    insmod: can't read 'xyz.ko': No such file or directory
    doing rmmod ...
    rmmod: chdir(/lib/modules): No such file or directory
    ONIE: Version       : 2015.9.14
    ONIE: Architecture  : x86_64
    ONIE: Machine       : accton_as5712_54x
    ONIE: Machine Rev   : 0
    ONIE: Config Version: 1
    Installing ONIE on: /dev/sda
    insmod: can't read 'xyz.ko': No such file or directory
    rmmod: chdir(/lib/modules): No such file or directory
    insmod: can't read 'xyz.ko': No such file or directory
    rmmod: chdir(/lib/modules): No such file or directory
    ONIE: Success: Firmware update URL: tftp://192.168.1.99/onie-updater-x86_64-accton_as5712_54x-r0
    ONIE: Success: Firmware update version: 2015.9.14
    ONIE: Rebooting...
    ONIE:/ # discover: Rescue mode detected. No discover stopped.
    Stopping: dropbear ssh daemon... done.
    Stopping: telnetd... done.
    Stopping: syslogd... done.
    Info: Unmounting kernel filesystems
    The system is going down NOW!
    Sent SIGTERM to all processes
    Sent SIGKILL toRestarting system.
    machine restart

As you can see, ``doing insmod ...`` and ``doing rmmod ...`` are displayed one time.  ``insmod`` and `` rmmod`` are executed 3 times.  ``xyz.ko`` is a nonexistent file in the test case.  (In our firmware update environment, the module is existent.)

I dowgraded ONIE to the previous version based on v2015.05 and everything looks find.

cbrune, 2015-09-15:
This is happening from commit a8e712b451dce0f40c8a9ce117922be758ba9a38.

The relevant part of the commit message:
```
    - update image information.  The update image format is an executable
      shell script archive file.  As such, the update image is a 'program'
      that can take parameters.  This patch adds the '-i' CLI parameter to
      the update image format, which is used to dump information about the
      update image.  When present this parameter will dump the
      machine.conf file contained within the update image, which includes
      items like build date, version string, target platform, etc.
```

The relevant source code is the x86 version of finish_update_install():
https://github.com/opencomputeproject/onie/blob/master/rootconf/x86_64/sysroot-lib-onie/exec-installer-arch#L33-L96

The new code "runs" the installer image with the arguments "-qi" in order to extract the image information.  This is done twice.

The code you added is failing before the command line argument code runs.

The command line argument parsing code, which parses the "-qi" arguments and emits the image information begins:

https://github.com/opencomputeproject/onie/blob/master/installer/install.sh#L74-L105

Generally speaking you should not have to touch `sharch_body.sh`, `install.sh` or `install-arch`.  If you have some reasons for wanting to touch these files let's discuss that.

`install.sh` sources `install-arch`, but that should not run any code directly. `install-arch` (and files it includes like `machine.conf` and `install.conf`) should only define variables and functions.  If that is not possible for what you want to do let's discuss that.

david56, 2015-09-16:
Thanks for your explanation.  I have understood the design.  The problem has been fixed.

closed 2015-09-16 by david56