opencomputeproject/onie #307: onie-2015.11.tar.gz build fail.

vic1231, 2015-12-30:
I'm trying to build newest release tar file, which is onie-2015.11.tar.gz.
But it will stop building for no reason.
Below is build log:
[INFO ]  Installing ltrace: done in 19.60s (at 30:57)
[INFO ]  =================================================================
[INFO ]  Installing strace
[INFO ]  Installing strace: done in 34.07s (at 31:31)
[INFO ]  =================================================================
[INFO ]  Cleaning-up the toolchain's directory
[INFO ]    Stripping all toolchain executables
[INFO ]  Cleaning-up the toolchain's directory: done in 3.72s (at 31:34)
[INFO ]  Build completed at 20151230.152300
[INFO ]  (elapsed: 31:33.87)
[INFO ]  Finishing installation (may take a few seconds)...
make[1]: Leaving directory '/home/wnc/project/onie-1230/onie/build/x-tools/x86_64-g4.9.2-lnx3.14.27-u0.9.33.2/build'
xyz@xyz-P243-M:~/project/onie/build-config$
xyz@xyz-P243-M:~/project/onie/build-config$
xyz@xyz-P243-M:~/project/onie/build-config$

Seems no error happen, 
After I build it again(), it will stop at other package build.

Below is build log:
gmknodes.o vgreduce.o vgremove.o vgrename.o vgscan.o vgsplit.o lvm.o \
	-llvm-internal -ldaemonclient  -ldl  -ldevmapper     -rdynamic
ar rs liblvm2cmd.a dumpconfig.o formats.o lvchange.o lvconvert.o lvcreate.o lvdisplay.o lvextend.o lvmchange.o lvmcmdline.o lvmdiskscan.o lvreduce.o lvremove.o lvrename.o lvresize.o lvscan.o polldaemon.o pvchange.o pvck.o pvcreate.o pvdisplay.o pvmove.o pvremove.o pvresize.o pvscan.o reporter.o segtypes.o toollib.o vgcfgbackup.o vgcfgrestore.o vgchange.o vgck.o vgcreate.o vgconvert.o vgdisplay.o vgexport.o vgextend.o vgimport.o vgmerge.o vgmknodes.o vgreduce.o vgremove.o vgrename.o vgscan.o vgsplit.o lvmcmdlib.o lvm2cmd.o
make[2]: Leaving directory '/home/wnc/project/onie-1230/onie/build/wnc_sst1_n1-r0/lvm2/lvm2-2_02_105/tools'
make -C daemons
make[2]: Entering directory '/home/wnc/project/onie-1230/onie/build/wnc_sst1_n1-r0/lvm2/lvm2-2_02_105/daemons'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/wnc/project/onie-1230/onie/build/wnc_sst1_n1-r0/lvm2/lvm2-2_02_105/daemons'
make[1]: Leaving directory '/home/wnc/project/onie-1230/onie/build/wnc_sst1_n1-r0/lvm2/lvm2-2_02_105'
xyz@xyz-P243-M:~/project/onie/build-config$
xyz@xyz-P243-M:~/project/onie/build-config$
xyz@xyz-P243-M:~/project/onie/build-config$

Where my build commands is:
make -j4 MACHINEROOT=../machine/wnc MACHINE=wnc_sst1_n1 all

cbrune, 2015-12-30:
The parallel make option "-j" is obscuring the actual failure.  Remove the "-j4" from your command and perform a fresh build after running "make distclean".  Also save the entire build output to a log file.

```
linux:$ cd build-config
linux:$ make distclean
linux:$ make MACHINEROOT=../machine/wnc MACHINE=wnc_sst1_n1 all 2>&1 | tee /tmp/build.log
```

That should stop with the error.  You also then have the build log to inspect.

vic1231, 2016-01-14:
Thanks, Curt.
I thought it's my build procedure problem, maybe modified something but do not remember.
After use clean package and build again, it works.

closed 2016-01-14 by vic1231