opencomputeproject/onie #310: crash in 'ip neigh'

bittorf, 2016-01-14:
see http://lists.busybox.net/pipermail/busybox/2016-January/083789.html

```

root at box:~ ip neigh show dev br-mastergate
Segmentation fault

but this works:

root at box:~ ip neigh show | grep 'br-mastergate'
192.168.100.111 dev br-mastergate lladdr e8:de:27:fb:ea:c9 used 0/0/0 probes 6 FAILED
192.168.2.104 dev br-mastergate lladdr 10:6f:3f:0e:31:8f used 0/0/0 probes 6 FAILED
10.35.50.168 dev br-mastergate lladdr 78:d6:f0:45:eb:55 used 0/0/0 probes 0 STALE
10.185.73.139 dev br-mastergate lladdr 78:d6:f0:45:eb:55 used 0/0/0 probes 0 STALE
169.254.123.93 dev br-mastergate lladdr 00:21:00:ce:c5:8f used 0/0/0 probes 0 STALE
192.168.178.37 dev br-mastergate lladdr c4:0a:cb:bc:04:70 used 0/0/0 probes 6 FAILED
192.168.23.75 dev br-mastergate lladdr 64:09:80:7a:e7:93 used 0/0/0 probes 6 FAILED
192.168.111.21 dev br-mastergate lladdr 64:70:02:d3:24:0b used 0/0/0 probes 6 FAILED
192.168.2.61 dev br-mastergate lladdr f4:ec:38:9d:86:36 ref 1 used 0/0/0 probes 4 INCOMPLETE
fe80::7ad6:f0ff:fe45:eb55 dev br-mastergate lladdr 78:d6:f0:45:eb:55 used 0/0/0 probes 0 STALE

also another 'real' interface (e.g. wlan1) works without issues,
the 'br-mastergate' is a bridge. (_maybe_ this is the culprit)
```

cbrune, 2016-01-14:
Hmmm, not sure what is going wrong here.

I can't reproduce it using an x86 system.  I created a bridge called `br-mastergate`.  The `ip neigh show dev br-mastergate` command works fine.

cbrune, 2016-01-14:
It seems the main difference between the two cases is when "dev br-mastergate" is specified to the query the following call is made:
```
	if (filter_dev)  {
		if ((G_filter.index = xll_name_to_index(filter_dev)) == 0) {
			bb_error_msg_and_die(bb_msg_invalid_arg,
					     filter_dev, "Cannot find device");
		}
	}
```

Could xll_name_to_index() be corrupting something....

In print_neigh() `G_filter.index` is checked in order to determine whether or not print the result.

bittorf, 2016-01-15:
i will debug this on the weekend directly on the mips-machine.

cbrune, 2018-08-06:
@bittorf - is this still an issue.  about to close issue as old business.

bittorf, 2018-08-22:
i do not think this is fixed. i will investigate during the next 30 day. please wait.