opencomputeproject/onie #329: [imt] [tests] mount: mounting none on /tmp/partFCfvBV/dev failed: No such file or directory

vitalivanov, 2016-02-02:
During ONIE testing on IMT HW the following error message was seen on screen.
Issue is seen only when NOS is installed.
As NOS we used ONL1 linux:
https://github.com/InterfaceMasters/ONL/tree/imt-devel-latest

If to perform NOS uninstall before the update then issue is not reproduced.

```
EXT3-fs (sda2): error: couldn't mount because of unsupported optional features (240)
EXT2-fs (sda2): error: couldn't mount because of unsupported optional features (240)
mount: mounting none on /tmp/partFCfvBV/dev failed: No such file or directory
umount: can't umount /tmp/partFCfvBV/sys: No such file or directory
umount: can't umount /tmp/partFCfvBV/proc: No such file or directory
umount: can't umount /tmp/partFCfvBV/boot: No such file or directory
umount: can't umount /tmp/partFCfvBV/dev: No such file or directory
mount: mounting none on /tmp/partFCfvBV/dev failed: No such file or directory
umount: can't umount /tmp/partFCfvBV/sys: No such file or directory
umount: can't umount /tmp/partFCfvBV/proc: No such file or directory
umount: can't umount /tmp/partFCfvBV/boot: No such file or directory
umount: can't umount /tmp/partFCfvBV/dev: No such file or directory
mount: mounting none on /tmp/partFCfvBV/dev failed: No such file or directory
umount: can't umount /tmp/partFCfvBV/sys: No such file or directory
umount: can't umount /tmp/partFCfvBV/proc: No such file or directory
umount: can't umount /tmp/partFCfvBV/boot: No such file or directory
umount: can't umount /tmp/partFCfvBV/dev: No such file or directory
ONIE: Success: Firmware update URL: http://192.168.157.56/onie-updater-x86_64-im_n29xx_t40n-r0 
ONIE: Success: Firmware update version: master-201601281358 
ONIE: Rebooting...
```

cbrune, 2017-06-07:
Is this still a problem?

Does the problem happen if the installed NOS is the ONIE DEMO OS?

vitalivanov, 2017-07-19:
@cbrune demo os is failing to install on imt hw #551

cbrune, 2017-08-21:
Closed by PR #589 and commit 31beb60.

closed 2017-08-21 by cbrune