opencomputeproject/onie #336: IPv6 neighbors are not showed after ping ff02::1 successfully

david56, 2016-03-02:
Hello @vidyaravipati, @cobracmder,

The issue should happen only in the Broadcom armv7a platforms (AS4610_54 and AS4610_30).

When doing neighbor discovery in `/bin/discover`, the IPv6 neighbors can't be showed after pinging ff02::1 successfully.

I created a simple function to reproduce issue:

    ONIE:/ # neigh_discovery()
    > {
    >     ping6 -I eth0 -c 2 ff02::1
    >     ping -I eth0 -w 2 -c 2 -q 255.255.255.255
    >     ip -4 neigh show
    >     ip -6 neigh show
    > }
    ONIE:/ # neigh_discovery
    PING ff02::1 (ff02::1): 56 data bytes
    64 bytes from fe80::ce37:abff:fe81:bcc6: seq=0 ttl=64 time=0.273 ms
    64 bytes from fe80::ce37:abff:fe81:bcc6: seq=1 ttl=64 time=0.190 ms
    
    --- ff02::1 ping statistics ---
    2 packets transmitted, 2 packets received, 0% packet loss
    round-trip min/avg/max = 0.190/0.231/0.273 ms
    PING 255.255.255.255 (255.255.255.255): 56 data bytes
    
    --- 255.255.255.255 ping statistics ---
    2 packets transmitted, 0 packets received, 100% packet loss
    ONIE:/ #

I performed `ip -6 neigh show` manually again, and the neighbor addr was showed.

    ONIE:/ # ip -6 neigh show
    fe80::ce37:abff:fe81:bcc6 dev eth0 lladdr cc:37:ab:81:bc:c6 ref 1 used 0/0/0 probes 1 REACHABLE
    ONIE:/ #

This addr will be aged out around 30 seconds.  So that ONIE never gathers the neighbors and adds them to the server list to discover images.

The behavior looks strange.   But it works fine in the x86_64 platforms such as AS7716_32X.  Do you have any idea about the issue?

david56, 2016-03-02:
BTW, the ONIE version is 2015.11.

cbrune, 2016-03-02:
Are you saying the neighbors eventually show up in the table?

The neigh_discovery() function does have a `sleep 4` (which I hate doing) after the pings and before the `ip neigh show` commands:
```
    # Allow neighbor table to populate -- better way?
    sleep 4
```

Are you saying that 4 seconds is not enough?  But a larger number does work?

I would like to better understand why it takes so long for the neighbor table to be populated....

cobracmder, 2016-03-02:
@david56 one of the things I ran into when running OCE on this platform last year was the machine running OCE wasn't configured properly so it was ignoring the IPv6 RA's and causing the IPv6 link local tests to fail.  Once I fixed that with sysctl (/etc/sysctl.conf), I got the behavior you just observed.  It wasn't until I rebooted everything: host and DUT did everything work the first time.

Can you verify if by doing this, is the problem resolved?

cbrune, 2016-03-02:
@david56 -- another thought.

Looking at the code here:
https://github.com/opencomputeproject/onie/blob/master/rootconf/default/bin/discover#L39-L50

The local IPv6 interface needs to leave the "tentative" state before the pings will work.  See https://tools.ietf.org/html/rfc2462:
```
   tentative address - an address whose uniqueness on a link is being
        verified, prior to its assignment to an interface.  A tentative
        address is not considered assigned to an interface in the usual
        sense. An interface discards received packets addressed to a
        tentative address, but accepts Neighbor Discovery packets
        related to Duplicate Address Detection for the tentative
        address.
```

Note the loop count on line 41 is set to `300`.  That waits for 30 seconds for the interface to leave the tentative state.  Also note there is no check to see if that timed out.  If it timed out the pings will not work.

Perhaps the arm platform (including the MAC and PHY) takes longer than 30 seconds for the interface to come up and leave the tentative state.  Just as a dumb test I would increase the 300 number, plus add a print statement and see what is happening.

david56, 2016-03-03:
@cbrune,

> The neigh_discovery() function does have a sleep 4 (which I hate doing) after the pings and before the ip neigh show commands:
> 
>     # Allow neighbor table to populate -- better way?
>     sleep 4

I'm sorry.  I missed the code.  I added `sleep 4` to the simple function and the neighbor table can be populated.  Yesterday, I observed that the table is populated instantly without `sleep 4` in x86_64 platform.  I misled armv7a platform should have the same behavior.

    ONIE:/ # cat /etc/machine.conf
    onie_version=2016.1.26-dirty
    onie_vendor_id=259
    onie_platform=x86_64-accton_as7716_32x-r0
    onie_machine=accton_as7716_32x
    onie_machine_rev=0
    onie_arch=x86_64
    onie_config_version=1
    onie_build_date="2016-01-27T19:35+0800"
    onie_partition_type=gpt
    onie_kernel_version=4.1.12
    onie_firmware=auto
    ONIE:/ # neigh_discovery()
    > {
    >     ping6 -I eth0 -c 2 ff02::1
    >     ping -I eth0 -w 2 -c 2 -q 255.255.255.255
    >     ip -4 neigh show
    >     ip -6 neigh show
    > }
    ONIE:/ # neigh_discovery
    PING ff02::1 (ff02::1): 56 data bytes
    64 bytes from fe80::7272:cfff:fe9c:1ac2: seq=0 ttl=64 time=0.313 ms
    64 bytes from fe80::7272:cfff:fe9c:1ac2: seq=1 ttl=64 time=0.134 ms
    
    --- ff02::1 ping statistics ---
    2 packets transmitted, 2 packets received, 0% packet loss
    round-trip min/avg/max = 0.134/0.223/0.313 ms
    PING 255.255.255.255 (255.255.255.255): 56 data bytes
    
    --- 255.255.255.255 ping statistics ---
    2 packets transmitted, 0 packets received, 100% packet loss
    fe80::7272:cfff:fe9c:1ac2 dev eth0 lladdr 70:72:cf:9c:1a:c2 ref 1 used 0/0/0 probes 0 DELAY
    ONIE:/ #

@cobracmder,

Thank you

> Can you verify if by doing this, is the problem resolved?
> 
> Here are the sysctl's that are needed (for IPv4, looking for my changes on IPv6):
> 
>     net.ipv4.icmp_echo_ignore_broadcasts = 0

Yes, it works fine.

    ONIE:/ # neigh_discovery()
    > {
    >     ping6 -I eth0 -c 2 ff02::1
    >     ping -I eth0 -w 2 -c 2 -q 255.255.255.255
    >     sleep 4
    >     ip -4 neigh show
    >     ip -6 neigh show
    > }
    ONIE:/ # neigh_discovery
    PING ff02::1 (ff02::1): 56 data bytes
    
    --- ff02::1 ping statistics ---
    2 packets transmitted, 0 packets received, 100% packet loss
    PING 255.255.255.255 (255.255.255.255): 56 data bytes
    
    --- 255.255.255.255 ping statistics ---
    2 packets transmitted, 2 packets received, 0% packet loss
    round-trip min/avg/max = 0.163/0.256/0.349 ms
    192.168.3.20 dev eth0 lladdr cc:37:ab:81:bc:c6 used 0/0/0 probes 0 STALE
    ONIE:/ #

I got another problem here.  After host booting, DUT performing `ping6 -I eth0 -c 2 ff02::1` has no responses from host.  If host pings DUT first, then DUT can ping host and get the responses.  Does it also need to tune the sysctl configs?

david56, 2016-03-04:
Hello @cbrune, @cobracmder,

> I got another problem here. After host booting, DUT performing `ping6 -I eth0 -c 2 ff02::1` has no responses from host. If host pings DUT first, then DUT can ping host and get the responses. Does it also need to tune the sysctl configs?

For the issue, we observed/compared the behaviors between AS4610_54 and AS5710_54X.  For each DUT, we caught the packets transceiving between host and DUT.  We found the DUT (AS4610_54) does not send "Neighbor Advertisement (Type 136)" packet to the host.  It cause the host did not send a ping-reply packet to the DUT.

Please see the logs caught by Wireshark in the attachment:

[as4610_54-as5710_54x-ipv6-neigh-packets.zip](https://github.com/opencomputeproject/onie/files/158292/as4610_54-as5710_54x-ipv6-neigh-packets.zip)

(The host IP is `fe80::222:cfff:fe25:1daa`)

cbrune, 2018-08-06:
@david56 - is this still a problem?

cbrune, 2018-09-21:
Closing inactive issue.

closed 2018-09-21 by cbrune