opencomputeproject/onie #338: Issue installing NOS image

suriya92, 2016-03-04:
Hi,

I'm new to ONIE. I would like to install an image which I have built. I'm using my local Apache http server to fetch the image, but I keep getting some random errors like this :

:** Installer Mode Enabled **
ONIE:/ #
ONIE:/ # install_url http://168.127.181.242/onie-installer.bin..
Stopping: discover... done.
Info: Fetching http://168.127.181.242/onie-installer.bin ...
Connecting to 168.127.181.242 (168.127.181.242:80)
installer            100% |*******************************|  6073k  0:00:00 ETA
ONIE: Executing installer: http://168.127.181.242/onie-installer.bin
/installer: line 2: ▒▒▒: not found
▒x▒^▒▒z▒^▒x▒^: not found^▒▒▒▒^▒닆0H▒▒▒!Á▒}▒▒▒@▒
▒@: not foundine 2:
/installer: line 2: ▒▒▒: not found
/installer: line 2: can't open ▒K▒Ѭfu▒n*}k▒▒▒▒R: no such file
/installer: line 2: ▒▒▒▒z▒5VlkK: not found
▒В^▒▒▒▒{x▒▒▒▒▒: not found
/installer: line 2: ▒▒@u
▒M▒: not found▒H!▒H▒▒}H▒▒H
/installer: line 9: syntax error: unexpected ")"

I tried both the bzI and vmlinux.bin image. And when  I tried to install a precompiled Ubuntu image which  I downloaded from ubuntu page, I get this : 

ONIE:/ # install_url http://168.127.181.242/onie-installer.deb2..
Stopping: discover... done.
Info: Fetching http://168.127.181.242/onie-installer.deb ...
Connecting to 168.127.181.242 (168.127.181.242:80)
installer            100% |*******************************| 39486k  0:00:00 ETA
ONIE: Executing installer: http://168.127.181.242/onie-installer.deb
/installer: line 1: syntax error: unexpected newline

Can somebody help me on this? I have been spending several days to figure out !

hanasu, 2016-03-04:
What is contained within your image? For example here's the output of a successful install on the Quanta LB9:

```
Package contains:
OS version lb9-r-1.4.12.00.img
Kernel version lb9-kernel-1.0.5.0.img
VM Tracer version vm-.img
```

suriya92, 2016-03-04:
The image is for a Linux kernel 4.4.3 which I have downloaded from kernel.org and built to get  bzi & vmlinux.bin. I also tried with a precompiled Ubuntu image (3.5.0-23-generic) for which I got the second error.

audihsu-qci, 2016-03-07:
http://opennetlinux.org/binaries
This is linux built for ONIE.

cbrune, 2016-03-07:
It looks like ONIE is working fine here, but there is a problem with the installer image you are using.

You should also ask this question on the ONL mailing list.

suriya92, 2016-03-07:
Thank you. I will reach out to them. But I was thinking any stable linux kernel image would work with ONIE.

cbrune, 2016-03-07:
The kernel is not the problem.  The problem lies in how the NOS installer is being built.  The installer image being used in this case does not look like an ONIE compatible installer.

suriya92, 2016-03-07:
Ah.Now I get it.And it seems like the precompiled ONL binaries doesn't support my platform( X86_64-redwood). Do you know how I can build a linux image this is ONIE-compatible?

cbrune, 2016-03-07:
The demonstration NOS that ships with ONIE is a good example:

https://github.com/opencomputeproject/onie/wiki/Demo-OS

suriya92, 2016-03-07:
Thank you so much !  I'll try that

suriya92, 2016-03-07:
And is there a way to enter the bootloader on a white box switch before onie grub menu pops up.  I would like to install a linux image through USB  on the switch.

suriya92, 2016-03-10:
Can anyone tell me how I can load an ubuntu/debian .iso /.bin image downloaded from internet through ONIE? It throws errors when I try to do so. Or do you have the ONIE-demo OS .bin image?

cbrune, 2016-03-10:
This describes how to convert a Debian ISO into an ONIE compatible installer:

https://github.com/opencomputeproject/onie/tree/master/contrib/debian-iso

Ubuntu or another distro would be similar, but not identical.

suriya92, 2016-03-10:
Thanks you ! I will try that

suriya92, 2016-03-10:
I tried that and i get : 
Verifying image checksum ... OK.
Preparing image archive ... OK
Loading new kernel ...
/tmp/tmp.igZQsv/installer/install.sh: line 18: kexec: not found.

Pls help me with your ideas..

cbrune, 2016-03-14:
Hmmm.  What version of ONIE are you running?  kexec should be available.

suriya92, 2016-03-14:
Version   : HEAD-201508120053-dirty
Is there a way to install kexec manually in ONIE?

suriya92, 2016-03-14:
Additionally, ONIE_config_version=1
onie_kernel_version=3.2.35

cbrune, 2017-06-07:
closing as inactive for over a year

closed 2017-06-07 by cbrune