opencomputeproject/onie #384: UBIFS support in ONIE

lakesh-sharma, 2016-06-14:
i see that busybox supports ubifs related tools. But i am not sure if it is fully supported in onie, or if some one has tried. I do see some ubifs related command steps in demo installers, so it should work.
Raising this 'issue' to get more info or possible fix for following 2 items.

(1) **minor number of ubi_ctrl is not correct :** 
i checked but could found if /dev/ubi_ctrl is being created by kernel itself or by onie startup scripts infra.
It seems that /dev/ubi_ctrl has minor number as 63, where as kernel sees it as 58. So essentially any access to this device driver would not go through.

ONIE:/ # cat /sys/devices/virtual/misc/ubi_ctrl/dev 
10:58
ONIE:/ # 
ONIE:/ # cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00400000 00100000 "uboot"
mtd1: 00400000 00100000 "uboot-env"
mtd2: 00200000 00100000 "shmoo"
mtd3: 00400000 00100000 "sys-eeprom"
mtd4: 02000000 00100000 "diags"
mtd5: 02000000 00100000 "onie"
mtd6: fb200000 00100000 "open"
ONIE:/ # ls -ltr /dev/ubi*
crw-rw----    1 root     0          10,  63 Jun  6  2016 /dev/ubi_ctrl
crw-rw----    1 root     0         253,   8 Jun  6  2016 /dev/ubi0_7
crw-rw----    1 root     0         253,   7 Jun  6  2016 /dev/ubi0_6
crw-rw----    1 root     0         253,   6 Jun  6  2016 /dev/ubi0_5
crw-rw----    1 root     0         253,   5 Jun  6  2016 /dev/ubi0_4
crw-rw----    1 root     0         253,   4 Jun  6  2016 /dev/ubi0_3
crw-rw----    1 root     0         253,   3 Jun  6  2016 /dev/ubi0_2
crw-rw----    1 root     0         253,   2 Jun  6  2016 /dev/ubi0_1
crw-rw----    1 root     0         253,   1 Jun  6  2016 /dev/ubi0_0
crw-rw----    1 root     0         253,   0 Jun  6  2016 /dev/ubi0
ONIE:/ # ls -ltr /dev/mtd*
lrwxrwxrwx    1 root     0                9 Jan  1 00:00 /dev/mtd-uboot-env -> /dev/mtd1
lrwxrwxrwx    1 root     0                9 Jan  1 00:00 /dev/mtd-uboot -> /dev/mtd0
lrwxrwxrwx    1 root     0                9 Jan  1 00:00 /dev/mtd-sys-eeprom -> /dev/mtd3
lrwxrwxrwx    1 root     0                9 Jan  1 00:00 /dev/mtd-shmoo -> /dev/mtd2
lrwxrwxrwx    1 root     0                9 Jan  1 00:00 /dev/mtd-open -> /dev/mtd6
lrwxrwxrwx    1 root     0                9 Jan  1 00:00 /dev/mtd-onie -> /dev/mtd5
lrwxrwxrwx    1 root     0                9 Jan  1 00:00 /dev/mtd-diags -> /dev/mtd4
brw-rw----    1 root     0          31,   7 Jun  6  2016 /dev/mtdblock7
brw-rw----    1 root     0          31,   6 Jun  6  2016 /dev/mtdblock6
brw-rw----    1 root     0          31,   5 Jun  6  2016 /dev/mtdblock5
brw-rw----    1 root     0          31,   4 Jun  6  2016 /dev/mtdblock4
brw-rw----    1 root     0          31,   3 Jun  6  2016 /dev/mtdblock3
brw-rw----    1 root     0          31,   2 Jun  6  2016 /dev/mtdblock2
brw-rw----    1 root     0          31,   1 Jun  6  2016 /dev/mtdblock1
brw-rw----    1 root     0          31,   0 Jun  6  2016 /dev/mtdblock0
crw-rw----    1 root     0          90,  14 Jun  6  2016 /dev/mtd7
crw-rw----    1 root     0          90,  12 Jun  6  2016 /dev/mtd6
crw-rw----    1 root     0          90,  10 Jun  6  2016 /dev/mtd5
crw-rw----    1 root     0          90,   8 Jun  6  2016 /dev/mtd4
crw-rw----    1 root     0          90,   6 Jun  6  2016 /dev/mtd3
crw-rw----    1 root     0          90,   4 Jun  6  2016 /dev/mtd2
crw-rw----    1 root     0          90,   2 Jun  6  2016 /dev/mtd1
crw-rw----    1 root     0          90,   0 Jun  6  2016 /dev/mtd0

ONIE:/ # cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00400000 00100000 "uboot"
mtd1: 00400000 00100000 "uboot-env"
mtd2: 00200000 00100000 "shmoo"
mtd3: 00400000 00100000 "sys-eeprom"
mtd4: 02000000 00100000 "diags"
mtd5: 02000000 00100000 "onie"
mtd6: fb200000 00100000 "open"
ONIE:/ #

lakesh-sharma, 2016-06-14:
(2) **ubifs volume creation does not work**:

Essentially i am going to try following commands to create an ubifs volume, i got these from a demo installer from onie github itself.

a). ubiformat  /dev/mtd4
b). ubiattach /dev/ubi_ctrl -m 4
c).  ubimkvol  /dev/ubi0 -N nos -s 33554432
d). mkdir /mnt/nos_boot/
e). mount -t ubifs -o sync ubi0:nos /mnt/nos_boot/

If all of these commands work as expected, then the user (or nos-installer ) can mount the volume
as in step(d), and copy the .itb to to its partition and then next time boot from it.

ONIE:/ # ubiformat  /dev/mtd6
ubiformat: mtd6 (nand), size 4213178368 bytes (3.9 GiB), 4018 eraseblocks of 1048576 bytes (1024.0 KiB), min. I/O size 8192 bytes
libscan: scanning eraseblock 4017 -- 100 % complete  
ubiformat: 3818 eraseblocks have valid erase counter, mean value is 3
ubiformat: 176 eraseblocks are supposedly empty
ubiformat: 8 bad eraseblocks found, numbers: 12, 13, 31, 39, 4014, 4015, 4016, 4017
ubiformat: warning!: 16 of 4010 eraseblocks contain non-ubifs data
ubiformat: continue? (yes/no)  yes
ubiformat: formatting eraseblock 4017 -- 100 % complete  
ONIE:/ # 
ONIE:/ # ubiattach /dev/ubi_ctrl -m 6
ubiattach: UBI_IOCATT: Inappropriate ioctl for device
ONIE:/ # 

ubiattach does not work , likely due to minor number issue (as listed in (1)), however if we manually
fix the minor number, then it does seem to work.

ONIE:/ # rm -rf /dev/ubi_ctrl
ONIE:/ # mknod /dev/ubi_ctrl c 10 58
ONIE:/ # ubiattach /dev/ubi_ctrl -m 6
[ 1759.767300] UBI: attaching mtd6 to ubi0
[ 1765.790410] UBI: scanning is finished
[ 1765.806552] UBI: attached mtd6 (name "open", size 4018 MiB) to ubi0
[ 1765.813027] UBI: PEB size: 1048576 bytes (1024 KiB), LEB size: 1032192 bytes
[ 1765.820299] UBI: min./max. I/O unit sizes: 8192/8192, sub-page size 8192
[ 1765.827213] UBI: VID header offset: 8192 (aligned 8192), data offset: 16384
[ 1765.834398] UBI: good PEBs: 4010, bad PEBs: 8, corrupted PEBs: 0
[ 1765.840595] UBI: user volume: 0, internal volumes: 1, max. volumes count: 128
[ 1765.847955] UBI: max/mean erase counter: 5/4, WL threshold: 4096, image sequence number: 1117948770
[ 1765.857292] UBI: available PEBs: 4006, total reserved PEBs: 4, PEBs reserved for bad PEB handling: 0
[ 1765.866727] UBI: background thread "ubi_bgt0d" started, PID 2211
ONIE:/ # 

Finally the volume creation fails as well.

ONIE:/ # ubimkvol  /dev/ubi0 -N nos -s 33554432
ubimkvol: UBI_IOCMKVOL: Inappropriate ioctl for device
ONIE:/ #

lakesh-sharma, 2016-06-14:
the logs were collected on 32bit ARM system, 4G NAND flash with cpu as iProc.
However similar can be reproduced on PPC as well.

cbrune, 2016-06-14:
There are a couple of PowerPC ONIE boxes that use ubifs.

See this machine's demo installer:
https://github.com/opencomputeproject/onie/blob/master/machine/centec/centec_e580_48x6q/demo/platform.conf

As for the "minor number of ubi_ctrl is not correct :" problem.  That is a real problem.  The build-config/scripts/make-devices.pl script makes that device at compile time.  It really needs to be detected and created at run time.

A better approach for that device would be to update rootconf/default/etc/init.d/makedev.sh to dynamically create the /dev/ubi_ctrl device node.  It is pretty straight forward just grep /proc/devices and call mknod accordingly.

Also read this email thread on this topic from a year ago:
http://lists.opencompute.org/pipermail/opencompute-onie/2015-April/000578.html

lakesh-sharma, 2016-06-14:
thanks for the info.
Originally i followed steps listed in onie/machine/centec/centec_e580_20q4z/demo/platform.conf
and i see that it is exactly same as the one you listed.

Yes, the minor number needs to be set at runtime, but in my case, even after re-creating the device node using mknod, the ubimkvol fails. So i am not sure if something is still missing in kernel or busybox for ubifs.

Or the device has to be created initially with correct minor, i.e. recreation of node with correct minor number is causing some issue.

lakesh-sharma, 2016-06-16:
issue reported here looks similar to the one reported in #82 (https://github.com/opencomputeproject/onie/issues/82).
i.e. the root cause could be same. i.e. mounting ubifs volume or creating one.

Can we pls update #82 for the issues high-lighted here and clone this to #82.

For ubifs commands, i would sent email to author of email you listed from onie mailer.
could be i am missing some usage param, i am just new to ubifs

cbrune, 2016-06-16:
I think your problem is the /dev/ubi0 nodes are also not correct.  These need to be made dynamically, like /dev/ubi_ctrl.

Something like this:
```
        # create ubi control node                                                                                                   
        ubictrl_nod=`cat /sys/class/misc/ubi_ctrl/dev | awk -F: '{print $1, $2}'`                                                   
        mknod -m 0660 /dev/ubi_ctrl c $ubictrl_nod                                                                                  
                                                                                                                                    
        # attach ubi                                                                                                                
        mtd_num=6                                                                                    
        ubiattach -m $mtd_num /dev/ubi_ctrl                                                                                         
                                                                                                                                    
        # create ubi nodes                                                                                                          
        ubi0_major=`cat /sys/class/ubi/ubi0/dev | awk -F: '{print $1}'`                                                             
        mknod -m 0660 /dev/ubi0 c $ubi0_major 0                                                                                     
        for ubi0_minor in `seq 1 8`; do                                                                                             
            mknod -m 0660 /dev/ubi0_$(($ubi0_minor - 1)) c $ubi0_major $ubi0_minor                                                  
        done                                                                                                                        
```

lakesh-sharma, 2016-06-16:
thanks, it does seem to work ok.
Do we need to append this snippet at makedev.sh

lakesh-sharma, 2016-06-16:
closing the issue as the sequence provided above  fixes it.

closed 2016-06-16 by lakesh-sharma