opencomputeproject/onie #403: onie-fwpkg: add operation prints unnecessary warning

cbrune, 2016-07-26:
The `onie-fwpkg add` operation prints an unnecessary warning message:

```
ONIE:/mnt/onie-boot/onie # onie-fwpkg add fw_updater.bin
Warning: No files exist for update:  fw_updater.bin
Staging firmware update:  fw_updater.bin                                                                                    
```

The issue is a common routine is used for both 'add' and 'remove'.  The common routine emits a warning if the file does not exist.  See the `remove_update()` function in onie-fwpkg:
https://github.com/opencomputeproject/onie/commit/477cd471415102466ee86226263eb4a5c23a9ec3#diff-03fe001f756be8a7558f78a49320ed1eR191

cbrune, 2016-07-26:
See thread:
http://lists.opencompute.org/pipermail/opencompute-onie/2016-July/001202.html

lakesh-sharma, 2016-07-26:
There is another minor issue, probably a documentation issue.

The description of onie-fwpkg command lists that  it would show the pending fw updates if invoked with no options. (https://github.com/opencomputeproject/onie/wiki/CLI-Reference#show-name----this-is-the-default-command) 

However implementation seems to be such that, it would do 'show' with no options.
i.e. show-pending and show-results.
https://github.com/opencomputeproject/onie/commit/477cd471415102466ee86226263eb4a5c23a9ec3#diff-03fe001f756be8a7558f78a49320ed1eR557

lakesh-sharma, 2016-08-05:
Following code changes fixes the issue. Essentially it is putting the code of remove_update inline in 'add' function, except the 'exist' check code. Pls check and if it is ok, then close the issue.

diff --git a/rootconf/x86_64/sysroot-bin/onie-fwpkg b/rootconf/x86_64/sysroot-bin/onie-fwpkg
index 583c62e..eafdd7c 100755
--- a/rootconf/x86_64/sysroot-bin/onie-fwpkg
+++ b/rootconf/x86_64/sysroot-bin/onie-fwpkg
@@ -233,7 +233,22 @@ cmd_add()
     # OK to stage the update.
 
     # Removing any old attempts and results files.
-    remove_update "$name"
+
+    # Removing any files related to the firmware update package
+    [ -f "${onie_update_pending_dir}/$name" ] && {
+        [ "$quiet" != "yes" ] && echo "Removing pending firmware update: $name"
+        rm -f $cmd_verbose "${onie_update_pending_dir}/$name"
+    }
+
+    [ -f "${onie_update_attempts_dir}/$name" ] && {
+        [ "$quiet" != "yes" ] && echo "Removing firmware update attempts status: $name"
+        rm -f $cmd_verbose "${onie_update_attempts_dir}/$name"
+    }
+
+    [ -f "${onie_update_results_dir}/$name" ] && {
+        [ "$quiet" != "yes" ] && echo "Removing firmware update results: $name"
+        rm -f $cmd_verbose "${onie_update_results_dir}/$name"
+    }
 
     # Stage the update
     [ "$quiet" != "yes" ] && echo "Staging firmware update: $file"

commit a50da2cd (2016-08-19): Add warn_exists argument in function remove_update

Additional argument warn_exists in function remove_update
would control the printing of warning message if file
does not exist.

Closes #403
closed 2016-08-19 by cbrune