opencomputeproject/onie #405: ONIE build issue on Windows 10 1607, Build 14393.10 (Windows Linux SubSystem) with Ubuntu 14.04

cobracmder, 2016-08-03:
Even though ONIE doesn't officially support Ubuntu build hosts (only latest stable Debian), I wanted to document this behavior here and cross link to the Windows Linux SubSystem folks:

To replicate:
```
$ cd onie/build-config
$ make MACHINE=kvm_x86_64 download
$ make MACHINE=kvm_x86_64 recovery-iso
```

Will result in the following error while building xtools:
```
[INFO ]  Installing libelf for the target
[ERROR]    configure: error: neither int nor long is 32-bit
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing libelf for the target'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_libelf_backend[scripts/build/companion_libs/200-libelf.sh@118]
[ERROR]  >>        called from: do_libelf_for_target[scripts/build/companion_libs/200-libelf.sh@80]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/home/los/working/onie/build/crosstool-ng/crosstool-ng-1.21.0/docs/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 64:44.70)
make[1]: *** [build] Error 1
make[1]: Leaving directory `/home/los/working/onie/build/x-tools/x86_64-g4.9.2-lnx4.1.28-u0.9.33.2/build'
make: *** [/home/los/working/onie/build/x-tools/x86_64-g4.9.2-lnx4.1.28-u0.9.33.2/stamp/xtools-build] Error 1
```

cobracmder, 2016-08-03:
Windows Feedback-Hub Link: feedback-hub:?contextid=677&feedbackid=525b2ed6-0578-4631-a23f-1cae95b8bd63&form=1&src=2

cbrune, 2018-08-06:
closing as old

closed 2018-08-06 by cbrune