opencomputeproject/onie #421: ERROR:ONIE: update failed

hungtsung, 2016-09-07:
Hi

I try to change linux kernel from 3.2.69 to 4.1.26. 

I use USB boot and select "ONIE: Embed ONIE" to install ONIE OS after the iso be built.

I got following message in the installing process:

Info: Found static url: file:///lib/onie/onie-updater
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ONIE: Unable to find 'Base MAC Address' TLV in EEPROM data.
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ONIE: Unable to find 'Serial Number' TLV in EEPROM data.
ONIE: Executing installer: file:///lib/onie/onie-updater
Verifying image checksum ... OK.
Preparing image archive ... OK.
ONIE: Version       : 2016.08-rc1.0.0.4-dirty
ONIE: Architecture  : x86_64
ONIE: Machine       : cel_seastone
ONIE: Machine Rev   : 0
ONIE: Config Version: 1
Installing ONIE on: /dev/sda
Found a dos partition table in /dev/sda2
[   20.126155] random: nonblocking pool is initialized
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
**ERROR:ONIE: update failed.
ONIE: ERROR: Firmware update URL: file:///lib/onie/onie-updater
ONIE: ERROR: Firmware update version: 2016.08-rc1.0.0.4-dirty
Failure: Unable to install image: file:///lib/onie/onie-updater**

========================================================

However, when I feel sad to reboot the machine and select mSATA (/dev/sda), the ONIE OS can work ...

so what's wrong in ONIE update and install

Thanks.

cbrune, 2016-09-07:
One of the last steps when installing ONIE is to update the EEPROM TLV 0x28 (Platform Name) and 0x29 (ONIE Version) with the data from the update image.  That step is failing here.  Hard to say from the info here if EEPROM access is completely busted or whether the data is just corrupt.

Suggestions to debug further:

1. try using the onie-syseeprom command manually from the command line.  For example can you write the 0x28 TLV?

```
ONIE:# onie-syseeprom -s 0x28="Testing Platform"
```

2. turn on verbose debug output in the installer.  In the `installer/x86_64/install-arch` file add `set -x` to the top of the `install_image()` function.

hungtsung, 2016-09-08:
Thank you for your suggestions. Following installing information is the original version. It is mean that I do not change anything.  It also got failed when update the EEPROM TLV. However, the ONIE is updated success that made me confuse. 

=================================================================
Info: Found static url: file:///lib/onie/onie-updater
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ONIE: Unable to find 'Base MAC Address' TLV in EEPROM data.
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ONIE: Unable to find 'Serial Number' TLV in EEPROM data.
ONIE: Executing installer: file:///lib/onie/onie-updater
Verifying image checksum ... OK.
Preparing image archive ... OK.
ONIE: Version       : master-201608151715.0.0.4-dirty
ONIE: Architecture  : x86_64
ONIE: Machine       : cel_seastone
ONIE: Machine Rev   : 0
ONIE: Config Version: 1
Installing ONIE on: /dev/sda
Found a dos partition table in /dev/sda2
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ONIE: Success: Firmware update URL: file:///lib/onie/onie-updater
ONIE: Success: Firmware update version: master-201608151715.0.0.4-dirty
ONIE: Rebooting...

==============================================================

Then I try your suggestion "onie-syseeprom -s 0x28="Testing Platform" " .  It work fine in original version as follow

=============================================================
ret = -1
ret = -1
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
Adding   TLV 0x28: Platform Name
ret = -1
ret = -1
Programming passed.
TlvInfo Header:
   Id String:    TlvInfo
   Version:      1
   Total Length: 24
TLV Name             Code Len Value

Platform Name        0x28  16 Testing Platform
CRC-32               0xFE   4 0x6993C196
Checksum is valid.

============================================================

but got ERROR in my version as follow:

=============================================================
ret = -1
write error
ERROR: Cannot open I2C device
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.

============================================================

So, I think that maybe the kernel patches are incorrect. It cause I2C to error. I will try your second suggestion to debug continuously. Let me know if you have any suggestions

Thanks again

lakesh-sharma, 2016-09-08:
can you tell as what is the CPU type ?
Also what is the onie kernel ver in both cases.

hungtsung, 2016-09-09:
I change linux kernel from 3.2.69 to 4.1.26.
Arch is x86_64
Actually, the 4.1.26 work fine but not in ONIE framework. 
Therefore, I just want to move it to ONIE.
Thanks

lakesh-sharma, 2016-09-09:
4.1.26 is not one of default onie kernels, and hence there are no additional patches for it.
Do you use 4.1.26 vanilla version in non-onie environment or with patches, i.e. the case, when you don't see  i2c errs.

cbrune, 2018-01-29:
Closed as inactive 1+ years

closed 2018-01-29 by cbrune