opencomputeproject/onie #428: Overlayfs support

eziegenbalg, 2016-10-10:
This is a request to add overlayfs support. This is useful if the installer mounts installed squashfs systems for post-install steps.

cbrune, 2016-10-10:
For the 4.1.y kernel series this is simply enabling `CONFIG_OVERLAY_FS` in the kernel configuration.

For example changing these kernel configs:
https://github.com/opencomputeproject/onie/tree/master/build-config/conf/kernel/4.1.28

https://github.com/opencomputeproject/onie/blob/master/build-config/conf/kernel/4.1.28/linux.armv7a.config#L960
https://github.com/opencomputeproject/onie/blob/master/build-config/conf/kernel/4.1.28/linux.powerpc-softfloat.config#L2130
https://github.com/opencomputeproject/onie/blob/master/build-config/conf/kernel/4.1.28/linux.x86_64.config#L2439

eziegenbalg, 2016-10-10:
Didn't see those. Thanks for the help!

closed 2016-10-10 by eziegenbalg