opencomputeproject/onie #435: Few enhancements to firmware update feature (multi-reboot, err-logging)

lakesh-sharma, 2016-10-25:
Opening this issue to get more info on some of items related to firmware update feature.

1. While BIOS and CPLD can be updated in a single step, followed by power-cycle. Some other HW components could require multi-stage update, i.e. control should come back to onie-update mode
after reboot , to continue updating the said HW component.
e.g lets say a HW component HW-COMP has 3 parts inside and each require a reboot for udpate.
HW-COMP consists of comp-a, comp-b and comp-c.
While it would not be an issue for uefi arch. But for legacy(MBR) case, the control goes to NOS after reboot. And there is no standard way for onie to gain control after reboot.
URL https://github.com/opencomputeproject/onie/wiki/Design-Spec-x86-NOS-Interface#x86-updating-and-embedding-onie
states about grub-reboot for NOS,. but we can't be sure that every NOS would implement, even so the one who implement, would have it available with same name of command.

So i am looking for the options to solve this issue for legacy platforms.
(a) should we manipulate MBR. I am not sure  as how to do it, neither it is recommended way to solve this issue.
(b) Manipulate NOS grub: it could achieve the desired goal, but then as each NOS grub env/config can be different from other, it actually forces to have different onie updater image for HW vendor.
Which is not a desirable outcome. Also it is possible that some NOS may not want their grub env/config to be altered.
(c) ONIE/OCP to enforce that every NOS should have a uniform API for onie/diag to stage next boot.

2. Have a logging function accessible to firmware update infra.
I see that log_fw_update is useful, but firmware infra code in machine/firmware would not have access to it.

cbrune, 2016-11-14:
I see.  I think option 'a' or a variant of that is the way to go.  In general, relying on the NOS to do anything special for firmware updates should be avoided.

For a multipart upgrade the `onie-fwpkg` and `onie-boot-mode` commands could be used at the end of each 'stage' to set the boot mode back to `upgrade` before the reboot.

We already added the hook for an updater that adds a `/tmp/reboot-cmd`.  That little script could set the boot mode to `upgrade`, position the next stage of the upgrade and reboot.

Imagine you had a firmware upgrade that contained the 3 comp-a, comp-b, comp-c.  The firmware upgrade logic could go like:
```
if comp-a needs installing:
   install comp-a
   onie-fwpkg add "myself"  <--- queue up the the entire upgrader again
   onie-boot-mode upgrade
   setup /tmp/reboot-cmd
   return success

if comp-b needs installing:
   install comp-b
   [ same as above ....]

if comp-c needs installing:
   install comp-c
   [ same as above ....]
```

The is the general idea.  I think all the pieces are there, it just needs to be fleshed out a bit.

The problematic thing is what if a comp-X upgrade fails for some reason.  The "partial upgrade" problem.  There is no good way to roll back.  Perhaps there can be a retry.

lakesh-sharma, 2016-11-14:
Setting next boot mode for onie is fine, but the critical issue here is to set the system to boot onie itself, on next boot, in case of legacy/MBR based system. System would boot to NOS by default in case of legacy/MBR mode.
So as i under-stand correctly, you are suggesting that we should try exploring the option (a), i.e. try to manipulate MBR. I am not sure as how it could be done, but i can think of following steps.
-Pack the default onie-only MBR dump(512 bytes) in onie fw image itself, and over-write it to system MBR at the end of update (using dd), if we need to do multi-reboot comp. update. Before this step, we need to take the backup of actual/running MBR of system and store it in onie's persistent mount location (/mnt/onie-boot/...). And also be sure to restore the actual MBR at the end of update.
Any interrupt(power-failure or otherwise) during MBR copy/restore step would require USB recovery.
So in my opinion, option (a) is bit risky. Not sure if all NOS vendor would like this approach.

Instead option (c) should had been the desired approach, as onie/ocp already recommends every NOS to provide next boot command. But i believe not all NOS vendors implement, even so the syntax/usage could be different.
Option (b) is still safe, but it would make HW vendors to built a specific/custom fw image for each NOS.

cbrune, 2016-11-15:
OK, yes I see the shared MBR is the real stinker here.

Before spending too much time trying to reason this out I have one big question:  Is this is a real world problem or is it hypothetical?

If this is only a "maybe, conceivable" problem then I suggest we skip it and spend time and energy moving to UEFI.

Also if this a real problem today, then there is not a lot we can do about it for the deployed boxes -- the ONIE on the boxes is what it is, same with the NOS.  If it is a real problem today then I think we just instruct the end customer how to apply 3 firmware updates in succession.

I do not think it is in our best interests to spend our time/energy polishing the legacy BIOS MBR methods.

However, going back to explore --

option C -- essentially this is requiring that from ONIE context we can call into a NOS.  this will never work in all cases.  For example a NOS might use file systems that ONIE does not support.  If this was enforced, then every NOS would be required to have at least one ONIE compatible partition with "stuff on it" that ONIE can read and execute.

option B -- similar non-starter.  While GRUB is very popular, NOSes could use LILO or ELILO or any loader that supports chainloading.  Again the GRUB config might be a on file system that ONIE's kernel does not understand.  And as you say too big of a burden on HW vendors.

option A - it is risky.  So is staging 3 firmware updates back-to-back.  There are a handful of operations in ONIE where if the power goes out at just the right time the box is a brick.  Your idea about saving and restoring the MBR is intriguing -- good idea.  That could go something like, all in ONIE context:
```
1. save current MBR (NOS's MBR) in a non-volatile location, like /mnt/onie-boot/onie/update
2. install GRUB (ONIE's GRUB goes into MBR)
3. install firmware updates
4. use onie-boot-mode to reboot into ONIE
5. reboot and repeat steps 3 and 4 as necessary
6. on the last stage copy the NOS's MBR back into place.
```

I think if this is *really* needed then something like option A is still the best bet.

Or just instruct folks how to install the 3 updates manually, which could be automated externally using orchestration tools like puppet, chef, ansible, etc.

lakesh-sharma, 2016-11-15:
Thanks for the info.
So in summary, the multi-reboot component update scenario on legacy/MBR system should be done manually on installed/deployed systems. However we can also pursue option-A and check if it is reliable and if it fixes the problem, although it is a risky approach.

closed 2016-11-15 by lakesh-sharma

lakesh-sharma, 2017-05-24:
i compared the MBR data after onie, diags and NOS install is done. MBR data i.e. Grub's boot.img i.e. 512 bytes remain same.
Hence we can conclude that just saving/replacing the MBR alone is not going to solve this issue.
Likely rest of grub images i.e. core.img or diskboot.img contains the logic of booting the NOS,diags,onie partition. Which in my opinion would be more complex to manipulate.