opencomputeproject/onie #449: Debug options for USB/embed mode install

lakesh-sharma, 2016-11-29:
In Typical USB embed mode install/recover based scenario, it is feasible that the management port is not up or not connected. Also it could be the case that the system-eeprom is not accessible or not programmed.
Following 2 scenarios for system-eeprom not being accessible case.
1. It is present but is empty , because ONIE is the first OS to be loaded, so user can program system-eeprom itself , once onie is loaded. OR can load diags using onie, and can program eeprom.
2. Accessibility issues: System-eeprom access itself needs to be debugged, so we need to have atleast onie or some OS running.

So question is, does onie install using usb embed mode, requires management port  to be up,
and system eeprom valid.
In logs below, we see that link is down and system eeprom is not accessible. I think , the management port link down should be fine and might not cause the usb embed to fail. But i am not sure about system-eeprom un-accessible case. The install err logs does not print any useful message, indicating that why it failed, is there any specific param in eeprom it was looking for.
I see that there is option to enable verbose mode. Do we need to rebuilt iso , after enabling verbose mode.

Please press Enter to activate this console. Info: eth0:  Checking link... down.
ONIE: eth0: link down.  Skipping configuration.
ONIE: Failed to configure eth0 interface
ONIE: Starting ONIE Service Discovery
Info: Found static url: file:///lib/onie/onie-updater
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ONIE: Unable to find 'Base MAC Address' TLV in EEPROM data.
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ONIE: Unable to find 'Serial Number' TLV in EEPROM data.
ONIE: Executing installer: file:///lib/onie/onie-updater
Verifying image checksum ... OK.
Preparing image archive ... OK.
ONIE: Version       :  <ONIE_VENDOR_VER>
ONIE: Architecture  : x86_64
ONIE: Machine       :  <MACHINE_NAME>
ONIE: Machine Rev   : 0
ONIE: Config Version: 1
Installing ONIE on: /dev/sda
random: nonblocking pool is initialized
Notice:  Invalid TLV header found.  Using default contents.
Notice:  Invalid TLV checksum found.  Using default contents.
ERROR:ONIE: update failed.
ONIE: ERROR: Firmware update URL: file:///lib/onie/onie-updater 
ONIE: ERROR: Firmware update version: >ONIE_VENDOR_VER> 
Failure: Unable to install image: file:///lib/onie/onie-updater

cbrune, 2016-12-14:
When the ONIE firmware is updated, by default it will attempt to update the EEPROM TLV:

https://github.com/opencomputeproject/onie/blob/master/installer/grub-arch/install-arch#L783-L786
https://github.com/opencomputeproject/onie/blob/master/installer/u-boot-arch/install-arch#L75-L78

Note, that code is the last line of the script, so that the exit code of the script is non-zero.  That then leads to the overall update failing.

I agree with you that failure to access the EEPROM should not make the install fail.  The fix looks straight forward.  Perhaps just tack `|| true` on to the end of the `onie-syseeprom` line.

lakesh-sharma, 2016-12-14:
thanks for the info.
let me try the changes suggested.

closed 2016-12-14 by lakesh-sharma