opencomputeproject/onie #460: Compilation Errors in compiling Kvm_x86_64 and qemu_armv8a

pankajbansal3073, 2016-12-23:
@cbrune Hello Curt,

i observed the compilation errors when try to build qemu_armv8a machine from master branch.
ERROR: Missing SYSROOT_LIB: ld-2.24.so
These are the errors for which i added lib64 changes in sysroot.make.
I think those changes are necessary for any 64 bit target architecture with the new Crosstool-ng.
Let us discuss how we can solve these compilation errors.

i have create a pull request with my changes again. i think these changes can be made more generic. maybe introduce a new variable for bitness of target architecture (32 bit or 64 bit)

or we can use crosstool gcc to find if __LP64__, _LP64 macros are defined or not? (https://gcc.gnu.org/onlinedocs/cpp/Common-Predefined-Macros.html#Common-Predefined-Macros)

what do you think ?

cbrune, 2017-01-05:
Seems to be specific to the armv8 as the x86_64 is building OK.

cbrune, 2017-03-29:
Closing as I believe this is fixed.

closed 2017-03-29 by cbrune