opencomputeproject/onie #466: FW-update: Staging from NOS after onie self update , 'pending' folder not present

lakesh-sharma, 2017-01-19:
FW update feature requires some dir(s) present in persistent area /mnt/onie-boot/onie/update.
Dir 'results' and 'update' itself are create during onie install or self-update by installer/x86_64/install-arch

However 'pending' dir would not get create until user boots to onie, which can happen after install/embed or self-update. As 'pending' dir is created by one of startup routines in rootconf/x86_64/sysroot-lib-onie/init-arch

One case could be that user just updated to newer version of onie (which supports fw update infra), using self-update, and then booted directly to NOS. i.e. user does not booted back to onie, which was needed for creating 'pending' folder.
Now if user likes to stage fw update from NOS, then user would encounter and error during onie-fwpkg invocation, as it would find 'pending' folder missing.

Just want to know if it is an expected behavior , or is it required for user to boot to onie atleast once,
after self-update, in order for platform/init etc.

cbrune, 2017-01-20:
Thanks for the great bug report, including lots of info.

The behavior sounds like a bug.

It looks like the `pending` directory should be created when the ONIE installer is run.

lakesh-sharma, 2017-01-20:
thanks for the info.
let me try the fix and then i would send the patch.

commit ee7a0d7e (2017-01-26): fw-update: create pending dir during onie install.

closes #466
closed 2017-01-26 by cbrune