opencomputeproject/onie #486: Unable to build GCC 4.9.2 based toolchain with GCC 6.X

sajones, 2017-02-15:
I am attempting to build ONIE on Fedora 24 using
```
make -j4 MACHINEROOT=../machine/quanta/ MACHINE=quanta_lb9 all
```
and it is failing with the following error:
```
[INFO ]  Installing final gcc compiler
[ERROR]    cfns.gperf:101:1: error: 'const char* libc_name_p(const char*, unsigned int)' redeclared inline with 'gnu_inline' attribute
[ERROR]    gmake[4]: *** [cp/except.o] Error 1
[ERROR]    gmake[3]: *** [all-gcc] Error 2
[ERROR]    gmake[2]: *** [all] Error 2
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing final gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@259]
[ERROR]  >>        called from: do_gcc_backend[scripts/build/cc/100-gcc.sh@1113]
[ERROR]  >>        called from: do_gcc_for_host[scripts/build/cc/100-gcc.sh@825]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc.sh@56]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@624]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/srv/buildroot/onie/build/crosstool-ng/crosstool-ng-11cb2ddd43fd1ff493f4b7cd63f1cf654294165f/docs/B - Known issues.txt'
```
From what I can tell, it is the same as in jcmvbkbc/crosstool-NG#24 but I am not sure of how the ONIE build system integrates and patches crosstool-NG. Have anyone else seen this before?

cbrune, 2017-02-15:
I agree, that does look like the same issue.

The ONIE build system applies patches to crosstool-NG from here:
https://github.com/opencomputeproject/onie/tree/master/patches/crosstool-NG

You can drop patches in there, update the series file and rebuild.

The commitment I made to the ONIE project is to make sure it always builds with the current stable version of Debian:
http://lists.opencompute.org/pipermail/opencompute-onie/2014-August/000063.html

The next version of Debian called "Stretch" is going to use gcc 6.3.0.  That is coming soon...

cbrune, 2017-05-16:
A goal this quarter is to get ONIE tuned up for building on a Debian 9 (code name stretch) system.  This issue is definitely on the list as debian stretch uses gcc 6.3.0.

byronyi, 2017-11-02:
I still met the same issue building the current master and the latest stable release (2017.08.1). I am using the current stable Debian 9:

```
$ uname -a
Linux ip-192-168-2-254 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_64 GNU/Linux
$ gcc --version
gcc (Debian 6.3.0-18) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The error log:

```
[INFO ]  =================================================================
[INFO ]  Installing final gcc compiler
[ERROR]    cfns.gperf:101:1: error: 'const char* libc_name_p(const char*, unsigned int)' redeclared inline with 'gnu_inline' attribute
[ERROR]    make[4]: *** [Makefile:1058: cp/except.o] Error 1
[ERROR]    make[4]: *** Waiting for unfinished jobs....
[ERROR]    make[3]: *** [Makefile:3976: all-gcc] Error 2
[ERROR]    make[2]: *** [Makefile:864: all] Error 2
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing final gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@338]
[ERROR]  >>        called from: do_gcc_backend[scripts/build/cc/100-gcc.sh@1145]
[ERROR]  >>        called from: do_gcc_for_host[scripts/build/cc/100-gcc.sh@832]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc.sh@56]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@653]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/data/onie/build/crosstool-ng/crosstool-ng-1.23.0/docs/B - Known issues.txt'
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]
[ERROR]  (elapsed: 12:05.59)
/data/onie/build/crosstool-ng/crosstool-ng-1.23.0/ct-ng:147: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/data/onie/build/x-tools/x86_64-g4.9.2-lnx3.10.0-54.0.1.el7.x86_64-uClibc-ng-1.0.22/build'
rm: cannot remove '/data/onie/build/x-tools/x86_64-g4.9.2-lnx3.10.0-54.0.1.el7.x86_64-uClibc-ng-1.0.22/build/.config.2': No such file or directory
make/xtools.make:123: recipe for target '/data/onie/build/x-tools/x86_64-g4.9.2-lnx3.10.0-54.0.1.el7.x86_64-uClibc-ng-1.0.22/stamp/xtools-build' failed
make: *** [/data/onie/build/x-tools/x86_64-g4.9.2-lnx3.10.0-54.0.1.el7.x86_64-uClibc-ng-1.0.22/stamp/xtools-build] Error 1
```

cbrune, 2017-11-02:
Indeed.  It appears that commit 4b887c2b0f793bbaf5b4b31791231123c1bd7e9b, which upgraded crosstools-NG and prepared for gcc-6.3.0 caused a regression for the gcc-4.9.2 build with host gcc-6.3.0.

Re-opening this.  The fix is straight forward.  It will be included in the 2017.11 release

byronyi, 2017-11-02:
Glad that we could catch up with the coming release. Thanks Curt!

commit 7aba9afa (2017-05-18): crosstool-ng: backport fix to compile gcc 4.9.2 using host gcc 6.3.0

Backport a crosstool-ng fix that allows compiling gcc 4.9.2 using gcc
6 host compiler.

  From 1e5f1089dec3af328fd03125d6778f666d0bd4e4 Mon Sep 17 00:00:00 2001
  From: edlinger <edlinger@138bc75d-0d04-0410-961f-82ee72b054a4>
  Date: Thu, 25 Feb 2016 15:33:50 +0000
  Subject: [PATCH 1/1] 2016-02-25  Bernd Edlinger  <bernd.edlinger@hotmail.de>

          Backported from mainline
          2016-02-19  Jakub Jelinek  <jakub@redhat.com>
                      Bernd Edlinger  <bernd.edlinger@hotmail.de>

          * Make-lang.in:  [... cut]
commit 01719c24 (2017-11-03): crosstool-ng: fix regression when compiling gcc 4.9.2 with host gcc 6.3.0

Commit 4b887c2b0f793bbaf5b4b31791231123c1bd7e9b caused a regression in
a previous fix for compiling gcc-4.9.2 with host gcc-6.3.0:

  commit 4b887c2b0f793bbaf5b4b31791231123c1bd7e9b
  Author: Curt Brune <curt@cumulusnetworks.com>
  Date:   Thu Jun 22 13:48:43 2017 -0700

      x86_64:upgrade toolchain to gcc-6.3.0 with uClibc-ng-1.0.22

The previous fix was from commit
7aba9afa04081c05ef3318de2eb6f2b3c947dcc4:

  commit 7aba9afa04081c05ef3318de2eb6f2b3c947dcc4
  Author: Curt Brune <curt@cumulusnetworks.com>
  Date:   We [... cut]
closed 2017-11-03 by cbrune