opencomputeproject/onie #487: Coreboot support is broken

cbrune, 2017-02-23:
I looks like some recent GRUB refactoring has broken the build for coreboot platforms.

See this thread:

http://lists.opencompute.org/pipermail/opencompute-onie/2017-February/001355.html

```
I'm having trouble compiling the Intel Rangeley recovery iso.

I keep getting an error message saying ERROR: Does not look like valid 
GRUB i386-pc library directory: 
onie-master/build/intel_coreboot_rangeley-r0/dev-sysroot/usr/lib/grub/i386-pc.
```

jf6316, 2017-06-14:
I have same error too. Please help to fix this bug.

reju85, 2017-06-19:
I also same error too. Please help to fix this bug.

cbrune, 2017-06-21:
Hi @jf6316  and @reju85  --

Thanks for pointing out the desire to have this fixed.

Can either of you help fix this problem?  We need someone with access to a coreboot setup to fix it.  I do not have one.

jf6316, 2017-07-31:
I have go back git branch "2017.02br" it's work fine thank you

donnydavis, 2017-11-28:
I built on Centos 7.4 and had to go back to 2016.11br

cbrune, 2017-11-29:
I made a PR that fixes the build.

However, I do not have a coreboot system, so have no way of verifying if the resulting image actually works or not.  @donnydavis, @reju85 or @jf6316  can you test it?

donnydavis, 2017-11-29:
I only have a generic C2758 machine to test with. I'm not sure if that is helpful

commit 156461b5 (2017-12-04): coreboot: build target grub-i386-pc for .ISO installer

In order to build an .ISO ONIE installer image we need the
grub-i386-pc components.

The ONIE build system builds the GRUB components twice.  Once using
the cross-compiler for the target architecture for use at runtime on
the target system.  And again using the host compiler for use by the
build system at build time.

On top of that, the cross-compiled components are built for a specific
firmware type: x86_64-efi, i386-pc or coreboot.

In ONIE versions 2016.11br and prior, the building of the .ISO ONIE
installer used the "host" version of [... cut]
closed 2017-12-04 by cbrune