opencomputeproject/onie #496: onie-uninstaller should stop the discovery process when run

cbrune, 2017-03-29:
Run `onie-uninstaller` while in onie-install mode. 20% through the process, discovery found an install image and rebooted the box, destroying GRUB.

The `onie-uninstall` command should stop the discovery process or fail with an error if you run it in install mode (as opposed to recovery mode).

david56, 2017-03-30:
> or fail with an error if you run it in install mode

The feature had been implemented in commit 9bdcbd6.  However, I think ONIE should stop auto discovery service and wait for the install/uninstall process accomplished before spawning a command-line shell for user.  That would keep user commands from fighting with ONIE services in the background.

cbrune, 2017-03-30:
Thanks for pointing out 9bdcbd6b9b0a -- this bug report must have been made against an older version of ONIE.

Commit 9bdcbd6b9b0a has been part of ONIE since release 2015.02.  Closing this.

closed 2017-03-30 by cbrune