opencomputeproject/onie #499: Unable to ping from ONIE environment

praveen0706, 2017-04-13:
Hi 

   I have downloaded the ONIE source and compiled for arm64 nxp-ls2088 board. When I load the onie linux, it has booted properly but the interface link status is showing NO. Due to this I am not able to assign the IP address for that, which is needed to get the ONL image.

  The interface is detected as ni0 , following errors are observed on the boot prompt.
[   1.959860] mdio_bus 8b96000: Error while reading PHY16 reg at 2.2
[    1.966116] mdio_bus 8b96000: Error while reading PHY16 reg at 3.3
[    1.972354] mdio_bus 8b96000: Error while reading PHY17 reg at 2.2
[    1.978597] mdio_bus 8b96000: Error while reading PHY17 reg at 3.3
[    1.984845] mdio_bus 8b96000: Error while reading PHY18 reg at 2.2
[    1.991082] mdio_bus 8b96000: Error while reading PHY18 reg at 3.3
[    1.997325] mdio_bus 8b96000: Error while reading PHY19 reg at 2.2
[    2.003569] mdio_bus 8b96000: Error while reading PHY19 reg at 3.3

/etc/rcS.d/S10networking.sh: line 213: arithmetic syntax error
Failure: Unable to configure MAC address for ni0
Info: Using ni0 MAC address:
Info: ni0:  Checking link... down.
ONIE: ni0: link down.  Skipping configuration.
ONIE: Failed to configure ni0 interface

The interface TX stats also not incremented when i do ping from the ONIE.

ONIE:/ # ifconfig  ni0 172.31.3.232 netmask 255.255.254.0 up
ONIE:/ # ethtool ni0
Settings for ni0:
        Supported ports: [ ]
        Supported link modes:   Not reported
        Supported pause frame use: No
        Supports auto-negotiation: No
        Advertised link modes:  Not reported
        Advertised pause frame use: No
        Advertised auto-negotiation: No
        Speed: Unknown!
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: off
        MDI-X: Unknown
        Link detected: no
ONIE:/ # ping 172.31.3.254
PING 172.31.3.254 (172.31.3.254): 56 data bytes

--- 172.31.3.254 ping statistics ---
4 packets transmitted, 0 packets received, 100% packet loss
ONIE:/ # ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:448 (448.0 B)  TX bytes:448 (448.0 B)

ni0       Link encap:Ethernet  HWaddr 36:DE:2F:0C:B6:C8
          inet addr:172.31.3.232  Bcast:172.31.3.255  Mask:255.255.254.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Do i need to do apply any patch OR any settings in the DPL/DPC file ?

Thanks,
Praveen D

cbrune, 2017-04-14:
Exactly which Accton board did you compile?

praveen0706, 2017-04-17:
Hi Cbrune,

  Able to resolve the issue, Iam using Accton 7711 board. We need to change the DPC file and generate .dtb file, then flash the new dtb in the corresponding address.

Thanks,
Praveen D

closed 2017-04-17 by praveen0706