opencomputeproject/onie #512: Installing ltrace error

digua88, 2017-05-04:
[INFO ]  Installing ltrace
**[ERROR]    plt.c:537:4: error: implicit declaration of function 'strndup' [-Werror=implicit-function-declaration]
[ERROR]    plt.c:537:21: error: incompatible implicit declaration of built-in function 'strndup' [-Werror]**
[ERROR]    make[6]: *** [plt.lo] Error 1
[ERROR]    make[5]: *** [all-recursive] Error 1
[ERROR]    make[4]: *** [all-recursive] Error 1
[ERROR]    make[3]: *** [all-recursive] Error 1
[ERROR]    make[2]: *** [all] Error 2
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing ltrace'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@259]

digua88, 2017-05-04:
strndup():
           Since glibc 2.10:
               _POSIX_C_SOURCE >= 200809L || _XOPEN_SOURCE >= 700
           Before glibc 2.10:
               _GNU_SOURCE

**How should I solve this problem?**

cbrune, 2017-05-04:
Hi -

You can reach a wider audience for help by sending an email to the ONIE mailing list:
http://lists.opencompute.org/mailman/listinfo/opencompute-onie

Can you please provide more information about the failure:

1. Send any patches you may have applied to the source tree
2. Send the `make ...` command you used to start the build
3. Details about the host build machine -- what Linux distro and version are you using?

I certify that the current ONIE release builds correctly on a Debian Jessie (version 8) build host.  Other distros, which may work fine, are not directly supported.  What tends to happen is later versions of GCC on some distros leads to incompatibilities.

Described here:  https://github.com/opencomputeproject/onie/wiki/Building-ONIE

Lastly -- if the build fails likes this while building the toolchain and you make some changes, say install a missing package or correct a typo in a configuration file, you should run `make distclean` to completely clean out the build area. Building the toolchain is a bit finicky.

digua88, 2017-05-08:
~/onie-2017.02_p1020/build$ ldd --version
ldd (Ubuntu EGLIBC 2.19-0ubuntu6.11) 2.19
$ gcc -v
gcc version 4.7.3 (Ubuntu/Linaro 4.7.3-12ubuntu1)

digua88, 2017-05-15:
--！

closed 2017-05-15 by digua88