opencomputeproject/onie #526: Any file saved is deleted automatically at reboot

adityakb95, 2017-06-02:
Any file created and saved in any directory including /home gets deleted on reboot. Can you please point out the solution to this

cbrune, 2017-06-02:
By design ONIE runs out of a ramdisk.  Generally it is not possible to save files across reboots.

On powerpc and other u-boot based machines some state is persisted in the U-Boot environment variables.

On x86_64 and other GRUB based machines some ONIE state is persisted in `/mnt/onie-boot/onie/config/`.

What are you trying to do?  What files do you want to store?

adityakb95, 2017-06-05:
I want to save some scripts(sh files) that automatically run on the machine
booting up

adityakb95, 2017-06-05:
I've managed to save the scripts permanently in the /mnt directory but can't figure out how to run them on boot up. Whatever script I try to save in init.d directory gets deleted on the next boot. Could you point me in the right direction. Any help will be greatly appreciated.

cbrune, 2017-06-05:
By design the end user can't modify the ONIE boot flow, like adding an init script.  The only way to add an init script is to rebuild and install an new version of ONIE.

What are you trying to do?  Perhaps there is another way to accomplish what you want to do.

The hardware manufacturer is response of the version of ONIE that ships on the box, as the ONIE init sequence contains machine specific details.

adityakb95, 2017-06-06:
Well I'm trying to run some inbuilt commands on each startup as I told you before. For that I found out that the startup scripts were in the init.d directory and tried to add my script and consequently my commands in the already stored scripts. Anyways as per your advice I'll contact the hardware manufacturer.

cbrune, 2017-08-31:
Closing as not a bug

closed 2017-08-31 by cbrune