opencomputeproject/onie #549: Gcc compiler for NOS

reju85, 2017-07-19:
Hi,
I am able compile and install ONIE and Nos images into accton_as7512_32x.It boots fine.For further development and validation i need to compile and install XDK images where i i need gcc compiler .I am not  able to  see gcc installed as part of NOS,but when downloading the source for ONIE  gcc 4.9 has been downloaded.How can i add gcc as a part of NOS to compile XDK images?

vitalivanov, 2017-07-19:
That's a question to NOS images vendor not ONIE.

reju85, 2017-07-19:
Hi,
The demo installer get generated as part of onie-master code.so is there any way that i can add gcc to demo OS ?

cbrune, 2017-07-19:
@reju85 - can you cross-compile the XDK (whatever that is) on the build host?

The cross-compiler is in `build/x-tools/x86_64-g6.3.0-lnx4.9.38-uClibc-ng-1.0.22/install/x86_64-onie-linux-uclibc/`, or something like that.

reju85, 2017-07-20:
HI,
Thank you for responding.Compiling with host PC will is possible but will not work  because the XDK has kernel module which we need to compile with onie kernel .where after installing onie on board we should copy and compile the module with the current  onie kernel so that kernel version matches.so for that i was asking how to add gcc as part of the demo OS in onie source

cbrune, 2017-07-20:
You should be able to do what you want on the build host.  You have the following pieces at your disposal:

1. the cross-compiler toolchain
2. the kernel source of the ONIE kernel
3. the build system / source for the XDK

You just need to tell the XDK build environment:
1. where the toolchain is
2. where kernel source is

How to do that with the XDK build environment is beyond my knowledge.  However, compiling a kernel module out of tree is a pretty common thing to do.  Here are some guidelines:
https://www.kernel.org/doc/Documentation/kbuild/modules.txt

For the particulars of the XDK, I would ask the XDK folks how to do that.

Trying to compile on the target machine is just asking for trouble.  What ONIE installs on the target machine is not suitable for development.

cbrune, 2017-08-31:
Closing as not a bug

closed 2017-08-31 by cbrune