opencomputeproject/onie #551: [imt] demo nos installation failure.

vitalivanov, 2017-07-19:
ONIE Demo NOS build as:
``
$ cd build-config
$ make -j4 MACHINEROOT=../machine/imt MACHINE=im_n29xx_t40n demo
``

Failure:
``
ONIE:/ # onie-nos-install http://192.168.157.164/onie/ONIE/demo-installer-x86_64
-im_n29xx_t40n-r0.bin
discover: installer mode detected.
Stopping: discover... done.
Info: Fetching http://192.168.157.164/onie/ONIE/demo-installer-x86_64-im_n29xx_t40n-r0.bin ...
Connecting to 192.168.157.164 (192.168.157.164:80)
installer            100% |*******************************|  9171k  0:00:00 ETA
ONIE: Executing installer: http://192.168.157.164/onie/ONIE/demo-installer-x86_64-im_n29xx_t40n-r0.bin
Verifying image checksum ... OK.
Preparing image archive ... OK.
Demo Installer: platform: x86_64-im_n29xx_t40n-r0
Creating new demo partition /dev/sda2 ...
mke2fs 1.42.8 (20-Jun-2013)
Discarding device blocks: done                            
Filesystem label=ONIE-DEMO-OS
OS type: Linux
Block size=1024 (log=0)
Fragment size=1024 (log=0)
Stride=0 blocks, Stripe width=0 blocks
32768 inodes, 131072 blocks
6553 blocks (5.00%) reserved for the super user
First data block=1
Maximum filesystem blocks=67371008
16 block groups
8192 blocks per group, 8192 fragments per group
2048 inodes per group
Superblock backups stored on blocks: 
        8193, 24577, 40961, 57345, 73729

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done 

Success: Support tarball created: /tmp/tmp.wnLJNU/onie-support.tar.bz2
/tmp/tmp.u5YdDp/installer/install.sh: .: line 341: can't open '/mnt/onie-boot/onie/grub/grub-variables'
Failure: Unable to install image: http://192.168.157.164/onie/ONIE/demo-installer-x86_64-im_n29xx_t40n-r0.bin
ONIE:/ #
``

cbrune, 2017-07-19:
I would think it has something to with the custom installer that `imt` is using:
https://github.com/opencomputeproject/onie/blob/master/machine/imt/im_n29xx_t40n/installer/install-platform

I've never tried to understand what the code is doing exactly.  Just assumed you folks knew what you wanted.

vitalivanov, 2017-07-20:
I will take care of it.

vitalivanov, 2017-08-23:
It installs and boots fine.

```
ONIE:/ # 
ONIE:/ # onie-nos-install http://172.16.0.56/onie/ONIE/demo-installer-x86_64-im_
n29xx_t40n-r0.bin
discover: installer mode detected.
Stopping: discover... done.
Info: Attempting http://172.16.0.56/onie/ONIE/demo-installer-x86_64-im_n29xx_t40n-r0.bin ...
Connecting to 172.16.0.56 (172.16.0.56:80)
installer            100% |*******************************|  9581k  0:00:00 ETA
ONIE: Executing installer: http://172.16.0.56/onie/ONIE/demo-installer-x86_64-im_n29xx_t40n-r0.bin
Verifying image checksum ... OK.
Preparing image archive ... OK.
Demo Installer: platform: x86_64-im_n29xx_t40n-r0
Creating new demo partition /dev/sda2 ...
mke2fs 1.42.13 (17-May-2015)
Discarding device blocks: done                            
Creating filesystem with 131072 1k blocks and 32768 inodes
Filesystem UUID: e158863e-cc66-414a-be43-3d59930b635c
Superblock backups stored on blocks: 
        8193, 24577, 40961, 57345, 73729

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done 

Success: Support tarball created: /tmp/tmp.9BEZqg/onie-support-im_n29xx_t40n.tar.bz2
Installing for i386-pc platform.
Installation finished. No error reported.
ONIE: NOS install successful: http://172.16.0.56/onie/ONIE/demo-installer-x86_64-im_n29xx_t40n-r0.bin
ONIE: Rebooting...
ONIE:/ # discover: installer mode detected.
Stopping: discover...start-stop-daemon: warning: killing process 397: No such process
 done.
Stopping: dropbear ssh daemon... done.
Stopping: telnetd... done.
Stopping: klogd... done.
Stopping: syslogd... done.
Info: Unmounting kernel filesystems
umount: devtmpfs busy - remounted read-only
umount: can't unmount /: Invalid argument
The system is going down NOW!
Sent SIGTERM to all processes
```

```
Welcome to the im_n29xx_t40n DEMO OS platform.

Please press Enter to activate this console. 
Welcome to the im_n29xx_t40n DEMO OS platform.
IM_N29XX_T40N-OS:/ # onie-sysinfo -a
Unknown Unknown 00:13:95:21:09:e7 master-201708230024-dirty 30324 im_n29xx_t40n 0 x86_64-im_n29xx_t40n-r0 x86_64 1 msdos 2m
IM_N29XX_T40N-OS:/ # uname -a
Linux onie 4.1.38-onie+ #1 SMP Wed Aug 23 01:06:54 EEST 2017 x86_64
IM_N29XX_T40N-OS:/ #
```

cbrune, 2017-08-23:
Looks like 13bd4da fixes this issue

closed 2017-08-23 by cbrune