opencomputeproject/onie #554: Build break in kernel 4.1.38

psarker, 2017-07-19:
Has anyone faced the build break for X86 platform in kernel 4.1.38 with gcc 4.9.2?
The same is observed in kernel 4.9.35 with gcc 6.3.0..

Checking for changes in the working directory ... Importing patch "onie-syseeprom-write-2b-delay-2000.patch" ... error: patch failed: miscutils/24cXX.c:44
error: miscutils/24cXX.c: patch does not apply
done

Any help will be greatly appreciated.

cbrune, 2017-07-19:
Hi,

Please send the exact command used to launch the build.

When was the last the machine built successfully?

cbrune, 2017-07-19:
Also note that the error you are seeing is during the patching of `busybox`, which has nothing to do with either the gcc version or the kernel version.

The busybox patches of the machine are not applying cleanly.

To verify you can do this:
```
linux:$ make MACHINEROOT=../machine/XYZ MACHINE=xyz clean
linux:$ make MACHINEROOT=../machine/XYZ MACHINE=xyz busybox-patch
```

psarker, 2017-07-19:
I was about to submit a new machine directory and today I have rebased my fork.. and I observed the break.. Yes, you are correct, it is not related to gcc version.. I did make distclean but still i have faced this .. ok, let me try as you have suggested....

psarker, 2017-07-19:
I remember last good build of my forthcoming platform I made was on 30th June or so..

psarker, 2017-07-20:
Please ignore this. My bad... The patch in busybox is no more required.. Platform was successfully compiled, tested and submitted for check in..

closed 2017-07-20 by psarker