opencomputeproject/onie #581: Make all fail on clean machine (git identity isn't set)

nshopik, 2017-08-17:
Git seems expect identity which isn't setup on clean installed debian here is failed log
```
Initialized empty Git repository in /home/shopik/onie-2017.08/build/crosstool-ng/crosstool-ng-1.23.0/.git/

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'shopik@MS-9618.(none)')
make/crosstool-ng.make:75: recipe for target '/home/shopik/onie-2017.08/build/crosstool-ng/stamp/crosstool-ng-patch' failed
make: *** [/home/shopik/onie-2017.08/build/crosstool-ng/stamp/crosstool-ng-patch] Error 128
make: *** Waiting for unfinished jobs....
```

Should we patch that or we expecting that it should be configured?

nshopik, 2017-08-17:
Looks like build is broken on debian testing, but it work fine on current debian stretch, could be debian issue, I didn't dig deeper
```
  GEN      doc/automake-1.15.1                                                                                                                                                                                     
help2man: can't get `--help' info from automake-1.15                                                                                                                                                               
Try `--no-discard-stderr' if option outputs to stderr                                                                                                                                                              
Makefile:3687: recipe for target 'doc/automake-1.15.1' failed           
```

cbrune, 2017-08-18:
Let's keep this bug about the empty git config...

The Debian stretch vs. testing is a non-issue for me.  It *is* a problem if building on Debian stretch is broken.

The git problem relates to using `stgit` and initializing the build trees that are patched with stgit.

I rate this problem as minor.

I suppose in a top level build step in the Makefile we could test the output of `git config --global user.email` and `git config --global user.name`, making sure that:
- the exit code `$?` is 0
- the contents are non-empty

nshopik, 2017-08-18:
> Let's keep this bug about the empty git config...

> The Debian stretch vs. testing is a non-issue for me. It is a problem if building on Debian stretch is broken.

Sorry about confusion, issue only appear on debian testing for some reason, maybe git had some changes in more recent versions? Debian stretch doesn't complain about empty git config nor debian jessie

cbrune, 2017-08-18:
OK, if this issue only happens on Debian testing then please close it.

nshopik, 2017-08-19:
Yeah, we probably may need it revisit it eventually if there some git changes made that this error throwing on newer versions

closed 2017-08-19 by nshopik