opencomputeproject/onie #584: automake fails to build

nshopik, 2017-08-18:
Looks like build is broken on debian testing, but it work fine on current debian stretch, could be debian issue, I didn't dig deeper
```
  GEN      doc/automake-1.15.1                                                                                                                                                                                     
help2man: can't get `--help' info from automake-1.15                                                                                                                                                               
Try `--no-discard-stderr' if option outputs to stderr                                                                                                                                                              
Makefile:3687: recipe for target 'doc/automake-1.15.1' failed           
```

cbrune, 2017-08-18:
OK, we will keep an eye on this for the next Debian release (Debian 10), codenamed buster.

nshopik, 2017-08-19:
Any pointers maybe I could debug it myself and find out real problem and report back with fix. My quick google fu doesn't seem give enough information what causing help2man fail in that case

cbrune, 2017-08-19:
The crosstool-ng log file is the place to start.  It is in:

`build/x-tools/x86_64-g6.3.0-lnx4.9.40-uClibc-ng-1.0.22/build/build.log` -- the exact xtool might vary based on the toolchain you are building.

cbrune, 2018-08-06:
@nshopik - is this still a problem?

nshopik, 2018-08-07:
I didn't build ONIE for while on debian 10, so I can't say for sure. Lets assume its fixed, and if its not I'll reopen

closed 2018-08-07 by nshopik