opencomputeproject/onie #585: mmcblk0 is not created

psarker, 2017-08-18:
Hi All, 
I am seeing a new problem on our PPC platforms.
/dev/mmcblk0 devnodes are not created after ONIE boots up in our platforms(lenovo/lenovo_ne1072t). Though it was good in past.
Is it possible that we need some tweak in order to make it work with devtmpfs?

Any help will be greatly appreciated.

Thanks,
Pinaki

cbrune, 2017-08-18:
1. Do you see the partitions in /sys/block/.... ?  If so, does 'mdev -s' help?

2. Does your kernel have devtmpfs support turned on?

3. Do you see the base device /dev/mmcblk0 ?  If your SD card is blank                                                
   (no partitions) that is all you would see.  You have to create                                                     
   partitions first, before they show up via devtmpfs.  You might need                                                
   to throw in a "partprobe /dev/mmcblk0" after partitioning for the                                                  
   kernel to see the new partitions.                                                                                  

4. Gross work around, manually create devices:                                                                        
```                                                                                                                      
  ONIE:# mknod /dev/mmcblk0 b 179 0                                                                                   
  ONIE:# for p in $(seq 7) ; do mknod /dev/mmcblk0p$p b 179 $p ; done                                                 
  ONIE:# ls -l /dev/mmc*                                                                                              
  brw-r--r--    1 root     0         179,   0 Aug 18 17:35 /dev/mmcblk0
  brw-r--r--    1 root     0         179,   1 Aug 18 17:34 /dev/mmcblk0p1                                             
  brw-r--r--    1 root     0         179,   2 Aug 18 17:34 /dev/mmcblk0p2                                             
  brw-r--r--    1 root     0         179,   3 Aug 18 17:34 /dev/mmcblk0p3                                             
  brw-r--r--    1 root     0         179,   4 Aug 18 17:34 /dev/mmcblk0p4                                             
  brw-r--r--    1 root     0         179,   5 Aug 18 17:34 /dev/mmcblk0p5                                             
  brw-r--r--    1 root     0         179,   6 Aug 18 17:34 /dev/mmcblk0p6                                             
  brw-r--r--    1 root     0         179,   7 Aug 18 17:34 /dev/mmcblk0p7                                             
```

What is the output of `cat /proc/partitions`, `ls /sys/block/*` and `ls -l /dev/*`.

cbrune, 2017-08-31:
These were fixed with ac1d6d5

closed 2017-08-31 by cbrune