opencomputeproject/onie #606: Unable to update ONIE due to changes in machine.conf naming

sonoble, 2017-09-19:
Hello,

In testing an issue reported to ONL about not installing on the latest ONIE master, I went to update one of my machines only to find that I cannot (easily) due to the name change.  I can obviously hack machine.conf to have the x86 vs msn2100 name, but I do not believe that is the proper solution.

ERROR:ONIE: Machine mismatch
Running platform     : x86_64-mlnx_msn2100-r0
Update Image platform: x86_64-mlnx_x86-r0
Source URL: http://onie-build/images/onie-updater-x86_64-mlnx_x86-r0

lakesh-sharma, 2017-09-19:
i believe onie already supports infra to fix this issue. Essentially you would need to over-ride the machine check function by defining it in machine/<your_onie_machine>/installer/install-platform as check_machine_image.
Pls check https://github.com/opencomputeproject/onie/search?p=1&q=check_machine_image&type=&utf8=%E2%9C%93
Above url shows usage of check_machine_image in various onie platforms.

sonoble, 2017-09-19:
I think I may be hitting a Mellanox issue, since I have not previously updated ONIE on one.  It appears that the Mellanox public ONIE only supports the MSX series platforms.

sonoble, 2017-09-19:
Closing as this is a reading comprehension issue.

closed 2017-09-19 by sonoble

cbrune, 2017-09-19:
Unfortunately the public mellanox onie is quite old.

Your best bet is to contact them directly.

byronyi, 2017-11-03:
I have met the same problem updating the ONIE image on MSN-2100.

This is the error log:

```
ONIE:/ # onie-nos-install http://192.168.2.254/onie-updater-x86_64-mlnx_x86-r0
Connecting to 192.168.2.254 (192.168.2.254:80)
Installer  100% | ************************ | 23451k.  0:00:00 ETA
ONIE: Executing installer: http://192.168.2.254/onie-updater-x86_64-mlnx_x86-r0
Verifying image checksum … OK.
Preparing image archive … OK.
ERROR:ONIE: Machine mismatch
Running platform	: x86_64-mlnx_x86-r5.0.1410
Update Image platform: x86_64-mlnx_x86-r0
Source URL: http://192.168.2.254/onie-updater-x86_64-mlnx_x86-r0
```

I am writing to Mellanox people and I will keep you posted.