opencomputeproject/onie #608: Build Process breaks for S6000 while building grub

fuu0, 2017-09-20:
Hey,

thanks for your work!
I try to build the Installation files for the Dell S6000, but it fails:
/home/username/downloads/onie-2017.08.01/build/user/x86_64-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/grub/grub-2.02~beta3/grub-core/osdep/unix/getroot.c:751:13: error: In the GNU C Library, "major" is defined by <sys/sysmacros.h>.

I uploaded the log here: https://github.com/opencomputeproject/onie/files/1318137/errors.txt)
Any ideas how I can fix that? I have no idea about C / C++ to debug that myself, sorry...

cbrune, 2017-09-20:
Thanks for the log file, very helpful.

It looks like the toolchain was not completely built.  This happens sometime when setting up a new build host -- the build breaks, you install some missing packages on the build host, try again, etc.  The build system gets a little out of sync thinking the toolchain was built OK when it wasn't.

I suggest starting fresh with a "make distclean" and then try building the dell-s6000 again.

BTW, what is `fmake` ?

fuu0, 2017-09-20:
Thanks for the quick reply! fmake was only an copy&paste error...
distclean changed it a bit - but I get still some errors.

Not the first error is this - and its not really helpful:
make[3]: *** [Makefile:5650: util/libgrubkern_a-getroot.o] Error 1
[errorlog.txt](https://github.com/opencomputeproject/onie/files/1318518/errorlog.txt)

Can you help me again?

cbrune, 2017-09-20:
The parallelism makes the output a little difficult to debug, but the compilation error is here:

```
home/username/downloads/onie-2017.08.01/build/user/x86_64-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/grub/grub-2.02~beta3/util/getroot.c: In function ‘grub_util_pull_device’:
/home/username/downloads/onie-2017.08.01/build/user/x86_64-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/grub/grub-2.02~beta3/util/getroot.c:101:7: error: this statement may fall through [-Werror=implicit-fallthrough=]
       grub_util_pull_lvm_by_command (os_dev);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/username/downloads/onie-2017.08.01/build/user/x86_64-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/grub/grub-2.02~beta3/util/getroot.c:103:5: note: here
     case GRUB_DEV_ABSTRACTION_LUKS:
     ^~~~
...
cc1: all warnings being treated as errors
make[3]: *** [Makefile:5650: util/libgrubkern_a-getroot.o] Error 1
make[3]: *** Waiting for unfinished jobs....
```

Compiling grub for the build host (using the build host's gcc) is failing.  We need some details about the build host:

- what distribution are you using?  what version?
- what is the version of gcc on the build host?

fuu0, 2017-09-20:
Fedora 26
Kernel 4.12.13
gcc version 7.1.1

cbrune, 2017-09-20:
Your version of gcc may be "too recent". Looks like it is detecting warnings in the grub code, which are interpreted as errors causing the build to fail.  Older versions of gcc are not detecting this warning.

I make sure the ONIE project builds with the current release of Debian (version 9 called Stretch).  The gcc version in stretch is 6.3.  Beyond that, getting ONIE to compile on all distros is best effort.

One option here is to fire up a virtual machine running Debian 9, or ubuntu-16.04 or ubuntu-17.04.

Another option is to patch the ONIE version of GRUB to fix the warning.

closed 2018-09-21 by cbrune