opencomputeproject/onie #647: Incremental build does not seem to work

lakesh-sharma, 2017-11-29:
Build time related optimizations had been enabled with following changes.
https://github.com/opencomputeproject/onie/pull/517

I tried this by pulling out latest onie work-space. And i like to highlight 2 issues.

(1) Incremental build does not seem to work with latest onie work-space.
By incremental build, what i meant is that some one build fresh onie images, and then does some minor changes in machine dir. e.g either enabling/disabling pxe. Or it could be updating installer.conf.
And then tries to build again. As these changes are minor, so user would expect that onie would rebuild
all images without rebuilding crosstool/grub etc. But instead it does nothing.
Listing the steps below.
(a) pull latest work-space.
(b) build s4000 images
make -j12 MACHINEROOT=../machine/dell MACHINE=dell_s4000_c2338 all
(c) images built fine. Now edited machine.make to disable pxe.
(d) tried to rebuild, but it does nothing.
make -j12 MACHINEROOT=../machine/dell MACHINE=dell_s4000_c2338 all

I tried same steps on 2016.08 base work-space, and it does seem to rebuild for step (d) above. And generates images within few minutes.
However it does so for one time, and would not rebuild afterwards.
But with latest work-space i.e. 2017.11 onwards, it does not rebuild at all.

Might be i am missing some steps, do i need to invoke 'make machine-clean' before re-build ?

cbrune, 2017-11-29:
As you point out I don't think the build has ever depended on `machine.make`.  That fact that it rebuilt once in 2016.08 is an accident, not by design.

In general it is difficult to know what needs rebuilding whenever `machine.make` is modified.  It could be anything: image assembly, add/remove components, installer changes, etc.

Whenever changing `machine.make` the thing to do is `make clean`, which is a synonym of `make machine-clean`.

Similar comments about `build-config/Makefile`.  At a minimum `make clean`, but depending on what changed might require `make distclean` first.

lakesh-sharma, 2017-11-29:
thanks for the info.
Closing this issue now.

closed 2017-11-29 by lakesh-sharma