opencomputeproject/onie #652: Adding support for Dell S4100 platform

lakesh-sharma, 2017-12-04:
[0001-Platform-support-for-the-DELL-EMC-S4100.zip](https://github.com/opencomputeproject/onie/files/1528444/0001-Platform-support-for-the-DELL-EMC-S4100.zip)

cbrune, 2017-12-04:
please make a pull request

lakesh-sharma, 2017-12-04:
I tried to raise pull req, by first committing to local branch and pushing the branch.
But got err, listing the steps below.

Pull the work-space
Created branch
   git branch dell_s4100
Switch to branch
  git checkout dell_s4100

Added Platform dir
Add and commit.
 git add dellemc_s4100_c2338

Tried to push the branch , so that pull req can be raised.

 git push origin dell_s4100
Username for 'https://github.com': lakesh-sharma
Password for 'https://lakesh-sharma@github.com': 
remote: Permission to opencomputeproject/onie.git denied to lakesh-sharma.
fatal: unable to access 'https://github.com/opencomputeproject/onie.git/': The requested URL returned error: 403

cbrune, 2017-12-04:
For pull requests, you first *clone* the repo.  Do all the work in your clone and push to your clone.  Then you make a pull request from your clone to the ONIE repo.  You never push to the ONIE repo.

See item 5 here: https://opencomputeproject.github.io/onie/developers/contributing.html#general-patching-philosophy

lakesh-sharma, 2017-12-04:
URL https://help.github.com/articles/creating-a-pull-request/
suggests creating branch
-------
These changes are proposed in a branch, which ensures that the master branch only contains finished and approved work.
------

do we need to create branch after cloning the repo i.e. using git-branch/git-checkout
or we need to create branch first at https://github.com/opencomputeproject/onie/pulls
and then clone the repo.

cbrune, 2017-12-05:
whoops! My bad, I used the wrong terminology.  First you *fork* the repo (not clone, by bad).

1. fork the repo into your github account
2. clone your fork to your workstation
3. in your fork create a branch  (git checkout -b <branch_name>)
4. do the work
5. push the branch to your fork
6. make the pull request (you should see a button on your github account to push at this point).

lakesh-sharma, 2017-12-05:
Thanks for updated steps.

I forked onie/ocp URL, and now it shows URL as https://github.com/lakesh-sharma/onie
Then i cloned the forked URL (  git clone https://github.com/lakesh-sharma/onie.git ).
However dir onie/machine/dell shows just 3 platforms, so not sure if forked and cloned work-space is not latest.
Also i see following message at url https://github.com/lakesh-sharma/onie
(
This branch is 2 commits ahead, 790 commits behind opencomputeproject:master.
)

Why forked/cloned work-space is not at latest.

cbrune, 2017-12-05:
Working with forks a bit strange, but ok once you are used to it.  The nuance is that the 'master' branch of the fork does *not* automatically stay in sync with the 'master' branch of the primary repo.  You have to update the master in the fork from time to time (whenever you want to work on a feature or bug).

Most important -- you never do actual work in your fork's master branch.  Only do work in a branch, branched off of the most recent master.

Here is what I do, based on this: https://help.github.com/articles/syncing-a-fork/

1. Add opencomputeproject/onie as a 'remote' to your fork:
```
# see current remotes
git remote -v
# add the primary repo as a remote -- calling it 'upstream' here, could be anything
git remote add upstream https://github.com/opencomputeproject/onie.git
# see update list of remotes
git remote -v
```

2. Sync your fork's master with the upstream master
```
# Check out your fork's local master branch.
git checkout master

# Fetch the branches and their respective commits from the upstream
# repository. Commits to master will be stored in a local branch,
# upstream/master.
git fetch upstream

# Merge the changes from upstream/master into your local master
# branch. This brings your fork's master branch into sync with the
# upstream repository, without losing your local changes.  This should
# be a fast-forward merge, i.e. identical to upstream
git merge upstream/master
```

3. Start work on a new feature/bug by creating a branch
```
git checkout -b feature-name
```

4. Push feature branch to your fork
```
git push origin feature-name
```

5. you should now see a pull-request button on your fork's homepage.

Whenever you want to work on a new feature repeat steps 2-5.  Step 1 only has to be done once.

lakesh-sharma, 2017-12-05:
thanks for detailed info and help.
Created pull req with #653

closed 2017-12-05 by lakesh-sharma