opencomputeproject/onie #656: ONIE: Rescue Mode  error

mikeleegit, 2017-12-11:
Hi 

I try to build the Installation files for the accton_as5712_54x,but it fails:

ONIE: Rescue Mode ...
error: fd0 cannot get C/H/S values.
error: fd0 cannot get C/H/S values.
Platform  :
Version   : HEAD-201712110940-dirty
Build Date: 2017-12-11T09:40+0800

Any ideas how I can fix that? I have no idea about C / C++ to debug that myself, sorry...

Centos 7 
kernel 3.10.0-693.5.2.el7.x86_64
gcc version 4.8.5 

Tks

cbrune, 2017-12-11:
@mikeleegit - can you describe your hardware setup in more detail?  What is the hardware platform exactly?

mikeleegit, 2017-12-12:
Hi

I try to build the Installation files for the accton_as5712_54x:Installed on usb,
This error message is displayed using usb-boot。
Do you want these messages?

The following is my computer hardware

processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 94
model name      : Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz
stepping        : 3
microcode       : 0x9e
cpu MHz         : 1392.875
cache size      : 6144 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 4
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 22
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch arat epb pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1
bogomips        : 6384.00
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 1
vendor_id       : GenuineIntel
cpu family      : 6
model           : 94
model name      : Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz
stepping        : 3
microcode       : 0x9e
cpu MHz         : 900.000
cache size      : 6144 KB
physical id     : 0
siblings        : 4
core id         : 1
cpu cores       : 4
apicid          : 2
initial apicid  : 2
fpu             : yes
fpu_exception   : yes
cpuid level     : 22
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch arat epb pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1
bogomips        : 6384.00
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 2
vendor_id       : GenuineIntel
cpu family      : 6
model           : 94
model name      : Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz
stepping        : 3
microcode       : 0x9e
cpu MHz         : 900.250
cache size      : 6144 KB
physical id     : 0
siblings        : 4
core id         : 2
cpu cores       : 4
apicid          : 4
initial apicid  : 4
fpu             : yes
fpu_exception   : yes
cpuid level     : 22
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch arat epb pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1
bogomips        : 6384.00
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 3
vendor_id       : GenuineIntel
cpu family      : 6
model           : 94
model name      : Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz
stepping        : 3
microcode       : 0x9e
cpu MHz         : 900.000
cache size      : 6144 KB
physical id     : 0
siblings        : 4
core id         : 3
cpu cores       : 4
apicid          : 6
initial apicid  : 6
fpu             : yes
fpu_exception   : yes
cpuid level     : 22
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch arat epb pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1

tks

david56, 2017-12-12:
@mikeleegit ,

Has you seen the console prompt message as below:

    Info: Mounting kernel filesystems... done.
    Info: Mounting ONIE-BOOT on /mnt/onie-boot ...
    Info: Using eth0 MAC address: cc:37:ab:f4:7f:94
    Info: eth0:  Checking link... up.
    Info: Trying DHCPv4 on interface: eth0
    ONIE: Using DHCPv4 addr: eth0: 192.168.1.200 / 255.255.255.0
    ONIE: Failed to configure eth1 interface
    ONIE: Failed to configure eth2 interface
    Starting: klogd... done.
    Starting: dropbear ssh daemon... done.
    Starting: telnetd... done.
    discover: Rescue mode detected.  Installer disabled.
    
    Please press Enter to activate this console.
    To check the install status inspect /var/log/onie.log.
    Try this:  tail -f /var/log/onie.log
    
    ** Rescue Mode Enabled **
    ONIE-RECOVERY:/ #

or hanging on the GRUB?

mikeleegit, 2017-12-12:
Hi 
I did not see these messages ~
It should be hanging on grub!

Because I click ONIE-Rescue will appear the following message
ONIE: Rescue Mode ...
GRUB loading. Error: fd0 can not get C / H / S values.
error: fd0 can not get C / H / S values.
Platform:
Version: HEAD-201712120938-dirty
Build Date: 2017-12-12T09: 38 + 0800

Press any key to continue ...
tks

david56, 2017-12-12:
Can you give me the serial number that should be printed in the bottom of AS5712_54X?  Thank you.

mikeleegit, 2017-12-12:
Hi 
Is that the serial number ?? Can be an example of it
tks

david56, 2017-12-12:
It would look like:

> SN: 5712XXXXXXXXXX

mikeleegit, 2017-12-13:
Hi

I do not have this serial number !! My current situation is the establishment of onie, the compiler accton_as5712_54x generated "onie-recovery-x86_64-accton_as5712_54x-r0.iso" Then this image on the computer using dd for usb,
Reboot this usb again, after the election usb-boot appeared these error message!
How should I deal with, or modify those code ??

tks

david56, 2017-12-13:
The serial number would help us to know detailed information (such as models of parts) to figure out the specification on the machine.  BTW, I do not see fd0 on AS5712_54X.