opencomputeproject/onie #664: ONIE install Debian in kvm_x86_64 error

lockless, 2017-12-15:
When installing debian in a kvm_x86_64, the following error occurs

  +---------------------+ Detecting network hardware +----------------------+
  |　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　 |
  |　　　　　　　　　　0%　 　　　　　　　　　　　　　　　　　　　　|
  |　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　 |
  | Detecting hardware, please wait... 　　　　　　　　　　　　　　　　　|
  |   　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　|
  +--------------------------------------------------------------------------------+

then waiting，but never continue...

Can I get some advice?

cbrune, 2017-12-15:
Hello @lockless -

A few questions.

- How did you launch the Debian installer from?  Can you share the screen output?
- What hypervisor are you using?

lockless, 2017-12-18:
Hi, thank you for your answer.
 The follow is the screen output:

ONIE:/ # onie-nos-install http://192.168.12.1/debian-jessie-amd64-mini-ONIE.bin
discover: Rescue mode detected. No discover stopped.
Info: Attempting http://192.168.12.1/debian-jessie-amd64-mini-ONIE.bin ...
Connecting to 192.168.12.1 (192.168.12.1:80)
installer            100% |*******************************| 17871k  0:00:00 ETA
ONIE: Executing installer: http://192.168.12.1/debian-jessie-amd64-mini-ONIE.bin
Verifying image checksum ... OK.
Preparing image archive ... OK.
Loading new kernel ...
my_load:667: do
kexec_core: Starting new kernel
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.16.0-4-amd64 (debian-kernel@lists.debian.org) (gcc version 4.8.4 (Debian 4.8.4-1) ) #1 SMP Debian 3.16.51-2 (2017-12-03)
[    0.000000] Command line: auto=true priority=critical  url=http://192.168.12.1/./debian-preseed.txt --- console=tty0 console=ttyS0,115200n8
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000100-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ee3bfff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ee3c000-0x000000003ee4dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003ee4e000-0x000000003fecdfff] usable
[    0.000000] BIOS-e820: [mem 0x000000003fece000-0x000000003ff25fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003ff26000-0x000000003ff2dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003ff2e000-0x000000003ff31fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003ff32000-0x000000003ffcffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ffd0000-0x000000003fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] DMI not present or invalid.
[    0.000000] Hypervisor detected: KVM
[    0.000000] AGP: No AGP bridge found
[    0.000000] e820: last_pfn = 0x3ffd0 max_arch_pfn = 0x400000000
[    0.000000] PAT not supported by CPU.
[    0.000000] init_memory_mapping: [mem 0x00000000-0x000fffff]
[    0.000000] init_memory_mapping: [mem 0x3ec00000-0x3edfffff]
[    0.000000] init_memory_mapping: [mem 0x3c000000-0x3ebfffff]
[    0.000000] init_memory_mapping: [mem 0x00100000-0x3bffffff]
[    0.000000] init_memory_mapping: [mem 0x3ee00000-0x3ee3bfff]
[    0.000000] init_memory_mapping: [mem 0x3ee4e000-0x3fecdfff]
[    0.000000] init_memory_mapping: [mem 0x3ff32000-0x3ffcffff]
[    0.000000] RAMDISK: [mem 0x3f05b000-0x3fecdfff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI BIOS Error (bug): A valid RSDP was not found (20140424/tbxfroot-212)
[    0.000000] No NUMA configuration found
[    0.000000] Faking a node at [mem 0x0000000000000000-0x000000003ffcffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x3ffcffff]
[    0.000000]   NODE_DATA [mem 0x3ffcb000-0x3ffcffff]
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 0:3ffc3001, primary cpu clock
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x00001000-0x00ffffff]
[    0.000000]   DMA32    [mem 0x01000000-0xffffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00001000-0x0009ffff]
[    0.000000]   node   0: [mem 0x00100000-0x3ee3bfff]
[    0.000000]   node   0: [mem 0x3ee4e000-0x3fecdfff]
[    0.000000]   node   0: [mem 0x3ff32000-0x3ffcffff]
[    0.000000] SFI: Simple Firmware Interface v0.81 http://simplefirmware.org
[    0.000000] smpboot: Allowing 1 CPUs, 0 hotplug CPUs
[    0.000000] PM: Registered nosave memory: [mem 0x000a0000-0x000fffff]
[    0.000000] PM: Registered nosave memory: [mem 0x3ee3c000-0x3ee4dfff]
[    0.000000] PM: Registered nosave memory: [mem 0x3fece000-0x3ff25fff]
[    0.000000] PM: Registered nosave memory: [mem 0x3ff26000-0x3ff2dfff]
[    0.000000] PM: Registered nosave memory: [mem 0x3ff2e000-0x3ff31fff]
[    0.000000] e820: [mem 0x40000000-0xffffffff] available for PCI devices
[    0.000000] Booting paravirtualized kernel on KVM
[... 114 lines cut ...]
[    0.751848] serio: i8042 KBD port at 0x60,0x64 irq 1
[    0.753071] serio: i8042 AUX port at 0x60,0x64 irq 12
[    0.754388] mousedev: PS/2 mouse device common for all mice
[    0.756131] input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio0/input/input0
[    0.758681] rtc_cmos rtc_cmos: rtc core: registered rtc_cmos as rtc0
[    0.760234] rtc_cmos rtc_cmos: alarms up to one day, 114 bytes nvram
[    0.761692] ledtrig-cpu: registered to indicate activity on CPUs
[    0.763089] AMD IOMMUv2 driver by Joerg Roedel <joerg.roedel@amd.com>
[    0.764602] AMD IOMMUv2 functionality not available on this system
[    0.766089] TCP: cubic registered
[    0.766951] NET: Registered protocol family 10
[    0.768112] mip6: Mobile IPv6
[    0.768855] NET: Registered protocol family 17
[    0.769867] mpls_gso: MPLS GSO support

+---------------------+ Detecting network hardware +----------------------+
|　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　 |
|　　　　　　　　　　0%　 　　　　　　　　　　　　　　　　　　　　|
|　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　 |
| Detecting hardware, please wait... 　　　　　　　　　　　　　　　　　|
| 　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　|
+--------------------------------------------------------------------------------+

to kill kvm:  sudo kill 132817
root@Ubuntu:/home/skunlin/onie-kvm#

cbrune, 2017-12-18:
Interesting.  I haven't tried this method in 2 years so I gave it a spin just now.

It works for me.  I know that does not help you much.

 I suspect there is something different in our VM configurations or perhaps even QEMU version.  May we can see what is different.

Here is what my console output looks like:
https://gist.github.com/cbrune/b809959115aa8ed8d3b4a053743824e1

I use a very vanilla qemu command line to start the vm.  Like this:
```
sudo qemu-system-x86_64 -enable-kvm -monitor stdio -name curt-onie -m 2048M -smp 2 -vnc 0.0.0.0:244 -vga std -serial telnet:localhost:9005,server -device e1000,netdev=onienet0 -netdev user,id=onienet0,hostfwd=:0.0.0.0:3040-:22,hostfwd=:0.0.0.0:3030-:3030 -drive file=/work/curt/trees/onie-cbrune/build/images/onie-recovery-x86_64-kvm_x86_64-r0.iso,media=cdrom -drive file=../onie-big-x86.qcow2,media=disk,if=virtio,index=0 -boot order=cd,once=d
```

This using qemu version 2.4.0 (old).

cbrune, 2017-12-18:
At the end it boot in Debian OK.  Here are some hardware details:
```
debian login: onie                 
Password:                          
Linux debian 3.16.0-4-amd64 #1 SMP Debian 3.16.51-2 (2017-12-03) x86_64                                                                     

The programs included with the Debian GNU/Linux system are free software;                                                                   
the exact distribution terms for each program are described in the    
individual files in /usr/share/doc/*/copyright.                       

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent     
permitted by applicable law.       
onie@debian:~$ uname -a            
Linux debian 3.16.0-4-amd64 #1 SMP Debian 3.16.51-2 (2017-12-03) x86_64 GNU/Linux                                                           
onie@debian:~$ sudo blkid
/dev/vda1: PARTLABEL="GRUB-BOOT" PARTUUID="f239fb30-130d-4bed-b188-f8e7dd4efcb5"
/dev/vda2: LABEL="ONIE-BOOT" UUID="69a8d8b8-c5f8-4e64-a6cb-bfe032d04172" TYPE="ext4" PTTYPE="dos" PARTLABEL="ONIE-BOOT" PARTUUID="a7969327-ce5d-42c2-9e58-bad3316b6edd"
/dev/vda3: UUID="1db0ecc0-1e1e-44ea-a817-66a6c3b311e3" TYPE="ext4" PARTUUID="1d203e23-1da6-4583-9de1-c584a54899bd"
/dev/vda4: UUID="7511e0a3-daa1-43e4-a52f-86fd2d0a8664" TYPE="swap" PARTUUID="9f5e8fa5-9d69-4cfa-af28-81004842b8fa"
```

lockless, 2017-12-19:
thank you so much, it seems that my vm has no nic.

lockless, 2017-12-19:
Thank you so much, it works now with your command. I guess the first my vm has no nic or it doesn't work.

closed 2017-12-19 by lockless

paulmenzel, 2021-04-13:
See [Installing Debian on a ONIE System](https://github.com/opencomputeproject/onie/tree/master/contrib/debian-iso) for reference.